InfiniTimeOrg/InfiniTime #2: build issue

maidenone, 2019-12-05:
```
[ 95%] Linking CXX executable pinetime-app.out
/home/maiden/Downloads/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/pinetime-app.dir/DisplayApp/DisplayApp.cpp.o: in function `Pinetime::Applications::DisplayApp::Process(void*)':
/home/maiden/Pinetime/src/DisplayApp/DisplayApp.cpp:78: undefined reference to `nrf_gfx_rect_draw(nrf_lcd_t const*, nrf_gfx_rect_t const*, unsigned short, unsigned long, bool)'
/home/maiden/Downloads/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /home/maiden/Pinetime/src/DisplayApp/DisplayApp.cpp:97: undefined reference to `nrf_gfx_print(nrf_lcd_t const*, nrf_gfx_point_t const*, unsigned short, char const*, FONT_INFO const*, bool)'
/home/maiden/Downloads/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/pinetime-app.dir/DisplayApp/DisplayApp.cpp.o: in function `Pinetime::Applications::DisplayApp::gfx_initialization()':
/home/maiden/Pinetime/src/DisplayApp/DisplayApp.cpp:156: undefined reference to `nrf_gfx_init(nrf_lcd_t const*)'
/home/maiden/Downloads/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /home/maiden/Pinetime/src/DisplayApp/DisplayApp.cpp:162: undefined reference to `nrf_gfx_rect_draw(nrf_lcd_t const*, nrf_gfx_rect_t const*, unsigned short, unsigned long, bool)'
/home/maiden/Downloads/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /home/maiden/Pinetime/src/DisplayApp/DisplayApp.cpp:177: undefined reference to `nrf_gfx_print(nrf_lcd_t const*, nrf_gfx_point_t const*, unsigned short, char const*, FONT_INFO const*, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/pinetime-app.dir/build.make:1381: src/pinetime-app.out] Error 1
make[1]: *** [CMakeFiles/Makefile2:210: src/CMakeFiles/pinetime-app.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

```

JF002, 2019-12-07:
I'm not sure why you have this error and not me, but in the last commit (https://github.com/JF002/Pinetime/commit/6fbb6c8f70b2103fd88d8d9da3ce884a283b1bfd), I've removed most of nrf_gfx. Could you give it a try?

maidenone, 2019-12-08:
Now it builds, but flashing fails (as i have nrfjprog in my syspath)

```
Scanning dependencies of target FLASH_pinetime-app
[ 96%] flashing pinetime-app.hex
/bin/sh: 1: /usr/bin/: Permission denied
make[2]: *** [src/CMakeFiles/FLASH_pinetime-app.dir/build.make:58: src/CMakeFiles/FLASH_pinetime-app] Error 126
make[1]: *** [CMakeFiles/Makefile2:102: src/CMakeFiles/FLASH_pinetime-app.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```
but i get the .out and .bin file :) and i can flash manually.

closed 2019-12-08 by maidenone