InfiniTimeOrg/InfiniTime #3: Make BLE softdevice optional

JF002, 2020-01-06:
Detect the presence of the BLE softdevice, and disable BLE if the softdevice is not present in memory.

This will allow the firmware to work even if the softdevice hasn't been uploaded to the device memory.

JF002, 2020-01-18:
This one doesn't seem easy, or even possible : when the MCU starts, the address of the first instructions it tries to execute correspond to an address that should be programmed by the softdevice.
If the softdevice is not flashed, there is not data in FLASH memory from 0x00000 to 0x26000, and the MCU executes garbage.

I think the only way to make the softdevice optional is to do it at compile time, but not at run time.

Any advice?

danielkucera, 2020-01-21:
For this purpose you could perhaps create a small bootloader that will be residing at 0x0 and will jump to the main app at 0x26000 to keep the main binary compatible between SD and non-SD version

JF002, 2020-01-21:
Good idea! I think I saw a reference to a MBR (Master Boot Record) in the NRF52-SDK. Maybe it could be useful.

JF002, 2020-06-16:
Since 0.5.0, the softdevice has been replaced by nimBLE as BLE stack. This issue is not relevant anymore.

closed 2020-06-16 by JF002