InfiniTimeOrg/InfiniTime #5: Support other programmer/debugger (OpenOCD, Black Magic Probe,...)

JF002, 2020-01-06:
Add the possibility to use other programmer/debugger than the JLink.
Ex : OpenOCD and Black  Magic Probe

JF002, 2020-01-17:
Support for BlackMagicProbe has been added in commit 81d629e86a888d41cd17c89a8c9c2dd6d596fb4e.

tgwaste, 2020-01-18:
Any advice on getting it to work using OpenOCD and Raspberry Pi?  Ive successfully compiled the app and have a pinetime-app.bin.  It looks like it flashes OK as well but nothing happens on the watch:

Open On-Chip Debugger 0.10.0+dev-01110-g8c459957 (2020-01-18-00:53)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
force hard breakpoints
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
spi_init spi mode: 80
bits per word: 8
max speed: 31200000 Hz (31200 KHz)
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
Stopping...
target halted due to breakpoint, current mode: Handler HardFault
xPSR: 0x21000003 pc: 0x0003718c msp: 0x2000ffc0

Flashing Application...
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00026c10 msp: 0x20010000
Enabled ARM Semihosting to show debug output
** Programming Started **
Info : nRF52832-QFAA(build code: E0) 512kB Flash, 64kB RAM
Warn : Adding extra erase range, 0x0002a22c .. 0x0002afff
** Programming Finished **
** Verify Started **
** Verified OK **

Restarting...
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00026c10 msp: 0x20010000, semihosting
Enabled ARM Semihosting to show debug output
target halted due to breakpoint, current mode: Handler HardFault
xPSR: 0x21000003 pc: 0x0003718c msp: 0x2000ffc0, semihosting

**** Done! Press Ctrl-C to exit...
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

JF002, 2020-01-18:
I've not implemented OpenOCD yet, but you can still flash the hex files manually.

Did you flash the ble softdevice in addition to the app? It must be flash at least once for the app to run correctly.
The file is available in the SDK (<sdk>/components/softdevice/s132/hex/s132_nrf52_6.1.1_softdevice.hex).

If you specify the path to the tool mergehex during CMake configuration (see README.md), the file pinetime-app-full.hex is generated. You could use it to program the pinetime for the first time.

tgwaste, 2020-01-18:
These tools are not available for ARM.

JF002, 2020-01-19:
Indeed, tools from NRF are x86 executables. I think it might be possible to merge the files without mergehex, but we'll have to understand how the two file are concatenated.

The addresses you used seem correct. Do you have to possibility to debug and see what happens in the soft (place breakpoints, run step by step,...)?

tgwaste, 2020-01-20:
I think I do with the new openocd-spi for the but honestly I just don't know how to turn it on and use it.

danielkucera, 2020-01-20:
@tgwaste how did you manage to unlock the device? The original wiki mentions using jlink but I only have stlink clone. I can connect to target but dap command doesn't work in that mode.

tgwaste, 2020-01-20:
I used OpenOCD and Raspberry Pi:
https://medium.com/@ly.lee/build-and-flash-rust-mynewt-firmware-for-pinetime-smart-watch-5e14259c55

tgwaste, 2020-01-20:
I just recently tried the pinetime-app-full.hex version but it still didnt work.
My script:
```
interface bcm2835gpio
bcm2835gpio_peripheral_base 0xFE000000
bcm2835gpio_speed_coeffs 236181 60
bcm2835gpio_swd_nums 20 21
bcm2835gpio_srst_num 18
transport select swd
source [find target/nrf52.cfg]
gdb_flash_program enable
gdb_breakpoint_override hard
$_TARGETNAME configure -event reset-init { arm semihosting enable }
init
echo ">>> Stopping"
reset
halt
echo ">>> Flashing Application"
program /root/pinetime-app-full.hex verify 0x00000000
echo ">>> Restarting"
reset init
resume
echo ">>> Flash Complete"
```
Ill work on trying to get debugging working.

danielkucera, 2020-01-20:
I'm also struggling to make it work: 
```
> program /home/danman/Projects/pinetime/Pinetime/build/src/pinetime-app.hex
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Adding extra erase range, 0x00050394 .. 0x00050fff
** Programming Finished **
> reset run
> 
> 
> reset run
> reset halt                                                                
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
> program /home/danman/Projects/pinetime/Pinetime/nRF5_SDK_15.3.0_59ac345/components/softdevice/s132/hex/s132_nrf52_6.1.1_softdevice.hex
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Padding image section 0 at 0x00000b00 with 1280 bytes
Adding extra erase range, 0x00025150 .. 0x00025fff
** Programming Finished **
> reset run
target halted due to breakpoint, current mode: Handler External Interrupt(22)
xPSR: 0x21070026 pc: 0x0002af42 msp: 0x2000fed0
> reset
target halted due to breakpoint, current mode: Handler External Interrupt(22)
xPSR: 0x21070026 pc: 0x0002af42 msp: 0x2000fed0
> 
> program /home/danman/Projects/pinetime/Pinetime/build/src/pinetime-app.hex                                                            
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000a80 msp: 0x20000400
** Programming Started **
Adding extra erase range, 0x00050394 .. 0x00050fff
** Programming Finished **
> reset
target halted due to breakpoint, current mode: Handler External Interrupt(22)
xPSR: 0x21070026 pc: 0x0002af42 msp: 0x2000fee0
```

danielkucera, 2020-01-20:
Ok, I managed to make it work. This is my upload script nrf52.cfg:
```
source [find interface/stlink.cfg]
transport select hla_swd
source [find target/nrf52.cfg]
init

reset halt
nrf51 mass_erase
flash write_image /home/danman/Projects/pinetime/Pinetime/build/src/pinetime-app.hex
flash write_image /home/danman/Projects/pinetime/Pinetime/nRF5_SDK_15.3.0_59ac345/components/softdevice/s132/hex/s132_nrf52_6.1.1_softdevice.hex 0

```
I run it like this:
```
/usr/local/bin/openocd -s tcl -f nrf52.cfg
```
after it starts, I connect via telnet and have to issue `reset run` several times until it stops saying `target halted`:
```
$ telnet localhost 4444
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> reset run
target halted due to breakpoint, current mode: Handler External Interrupt(22)
xPSR: 0x21070026 pc: 0x0002af42 msp: 0x2000fed0
> reset run
target halted due to breakpoint, current mode: Handler External Interrupt(22)
xPSR: 0x21070026 pc: 0x0002af42 msp: 0x2000fef0
> reset run
target halted due to breakpoint, current mode: Handler External Interrupt(22)
xPSR: 0x21070026 pc: 0x0002af42 msp: 0x2000fee0
> reset run
```

danielkucera, 2020-01-20:
[image]

tgwaste, 2020-01-20:
The reset run worked for me as well:

[image]

JF002, 2020-01-21:
Nice! I'm glad you got it working!
But I don't know why you have to issue several reset to make it work...

lupyuen, 2020-01-26:
This sounds like the unreliable GPIO Bit Bang issue that I fixed with openocd-spi? Try this... https://medium.com/@ly.lee/openocd-on-raspberry-pi-better-with-swd-on-spi-7dea9caeb590?source=friends_link&sk=df399bfd913d3e262447d28aa5af6b63

tgwaste, 2020-01-26:
I’ve been using openocd-spi since the first day you released it :)

lupyuen, 2020-01-26:
Haha super cool! :-) Yes there will be bugs with openocd-spi... let's eradicate them!

closed 2020-10-04 by JF002