InfiniTimeOrg/InfiniTime #8: Improve documentation about building and programming the application

JF002, 2020-01-06:
Add more info on building and programming the application into the PineTime (more details, commands, examples) for different environnment (clion, command line,...).

JF002, 2020-01-17:
I've added documentation about building and programming using JLink and BlackMagicProbe in commit 81d629e86a888d41cd17c89a8c9c2dd6d596fb4e.
Let me know if more information is needed!

alastair-dm, 2020-01-23:
Something about [openocd](https://github.com/lupyuen/openocd-spi) would be good for those without the above, but with a Pi.

JF002, 2020-01-26:
[This part of the pinetime wiki](https://wiki.pine64.org/index.php/PineTime#Using_OpenOCD_and_a_Raspberry_Pi) and [this article ](https://medium.com/@ly.lee/openocd-on-raspberry-pi-better-with-swd-on-spi-7dea9caeb590) might be useful. 
I don't use Openocd, so I can't write specific documentation about it, but I can add those links in README if they are helpful.

alastair-dm, 2020-01-27:
Having dug a little deeper I see the USE_OPENOCD option uses a hardcoded interface/stlink.cfg on the openocd command line, without a config option to change it, so it'll need more than a change in documentation to make it work. I'll probably create a new issue for it. For documentation, now that I can see what it's doing to flash I'll see if I can flash manually using openocd-spi and the binaries from the 0.2.0 release.

JF002, 2020-01-27:
Mhm maybe this file was missing during the pull-request? See https://github.com/JF002/Pinetime/pull/20

alastair-dm, 2020-01-28:
Pull request #20 adds the code I was looking at, which includes in its COMMAND lines the section:
`-f interface/stlink.cfg -c 'transport select hla_swd'`
That's correct (I assume) for the STlink v2 clones it was intended for, but wrong for either the Pi's GPIO or SPI which need a different interface and probably other bits in the command. I guess it'll end up needing a new OPENOCD_INTERFACE option, but we need to work out exactly what's needed in the command for the Pi. Further discussion probably belongs in issue #5 where various 'works for me' commands have been exchanged, and an issue with breakpoints discussed. Until that's sorted the best documentation change for flashing with the Pi would be to say the Pi isn't supported yet, but look at #5 for possible ways to flash manually. For me the missing link was the 'reset run' to get past the breakpoint.

incyi, 2020-02-14:
I can make a .travis.yml file to do automatic builds on travis-ci. Is this something what you want ?

JF002, 2020-02-15:
I don't use Travis for now, but I'm thinking about using a CI for the release builds, so it could be useful one day or another ;)

closed 2020-10-04 by JF002