InfiniTimeOrg/InfiniTime #12: Touch panel "crashes" when issuing sleep command

JF002, 2020-01-12:
CST816S seems to support a SLEEP command.
When I tried to call it, the controller wouldn't send me actual touch point values, it would send only static/erratic values.

As we don't have much documentation about this controller, this bug is not easy to diagnose.

I hope we'll get more info about CST816S soon so that we can improve the implementation.

JF002, 2020-04-04:
I was sending the wrong value to the register (0x05 instead of 0x03). The pinetime wiki has been updated : https://wiki.pine64.org/index.php/PineTime#Touch_events.

Now, we can try to reimplement the SLEEP command as detailed in the wiki and see if it works.

endian-albin, 2020-07-13:
I know the ST7789V display supports sleep mode (I've implemented it myself in the upstream Zephyr driver), but you're saying that the touch controller has a sleep mode of its own? When the display is in sleep mode it has close to zero power consumption.

sdorre, 2020-07-17:
The datasheet mentions this:

http://files.pine64.org/doc/datasheet/pinetime/CST816S%E6%95%B0%E6%8D%AE%E6%89%8B%E5%86%8CV1.1.pdf
```
4.5 Low Power Mode
The CST816S touch controller supports the following low-power modes:

Sleep mode: after the host sends a sleep command to the chip, the chip will immediately enter a deep sleep mode to achieve the lowest power consumption;
Through the Reset, the chip will wake up and enter a dynamic operating mode.

Standby mode: In this mode, the chip remains at a low frequency for a minimum scan to match a predefined wake-up gesture.

Translated with www.DeepL.com/Translator (free version)
```

So the behaviour you observed is "okay", i would say. Maybe only send this command when you go into sleep.
And when you wake up, trigger a reset and you are ready to go again.

Typical power consumption < 2.5mA in dynamic mode.
Typical power consumption < 10uA in standby mode.
Typical power consumption in sleep mode < 5uA;

JF002, 2020-09-20:
The correct sleep command has been implemented and works perfectly !

closed 2020-09-20 by JF002