InfiniTimeOrg/InfiniTime #13: make[2]: *** [src/CMakeFiles/FLASH_ERASE.dir/build.make:58: src/CMakeFiles/FLASH_ERASE] Error 33

ishotjr, 2020-01-15:
I'm running into the following when I try to `make -j` with `6abf12f`:

```bash
[  0%] erasing flashing
[  1%] started JLink commands
[  2%] flashing SoftDevice
[ 99%] Built target pinetime-app
[100%] flashing pinetime-app.hex
# _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
# _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
# watch_fast: "/org/gnome/terminal/legacy/" (establishing: 0, active: 0)
# unwatch_fast: "/org/gnome/terminal/legacy/" (active: 0, establishing: 1)
# watch_established: "/org/gnome/terminal/legacy/" (establishing: 0)
# _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
# _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
# watch_fast: "/org/gnome/terminal/legacy/" (establishing: 0, active: 0)
# unwatch_fast: "/org/gnome/terminal/legacy/" (active: 0, establishing: 1)
# watch_established: "/org/gnome/terminal/legacy/" (establishing: 0)
ERROR: JLinkARM DLL reported an error. Try again. If error condition
ERROR: persists, run the same command again with argument --log, contact Nordic
ERROR: Semiconductor and provide the generated log.log file to them.
make[2]: *** [src/CMakeFiles/FLASH_ERASE.dir/build.make:58: src/CMakeFiles/FLASH_ERASE] Error 33
make[1]: *** [CMakeFiles/Makefile2:156: src/CMakeFiles/FLASH_ERASE.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Parsing hex file.
Parsing hex file.
Erasing page at address 0x0.
Erasing page at address 0x1000.
Erasing page at address 0x2000.
Erasing page at address 0x3000.
Erasing page at address 0x4000.
Erasing page at address 0x5000.
Erasing page at address 0x6000.
Erasing page at address 0x7000.
Erasing page at address 0x8000.
Erasing page at address 0x9000.
Erasing page at address 0xA000.
Erasing page at address 0xB000.
Erasing page at address 0xC000.
Erasing page at address 0xD000.
Erasing page at address 0xE000.
Erasing page at address 0xF000.
Erasing page at address 0x10000.
Erasing page at address 0x11000.
Erasing page at address 0x12000.
Erasing page at address 0x13000.
Erasing page at address 0x14000.
Erasing page at address 0x15000.
Erasing page at address 0x16000.
ERROR: JLinkARM DLL reported an error. Try again. If error condition
ERROR: persists, run the same command again with argument --log, contact Nordic
ERROR: Semiconductor and provide the generated log.log file to them.
make[2]: *** [src/CMakeFiles/FLASH_pinetime-app.dir/build.make:58: src/CMakeFiles/FLASH_pinetime-app] Error 33
Erasing page at address 0x17000.
make[1]: *** [CMakeFiles/Makefile2:102: src/CMakeFiles/FLASH_pinetime-app.dir/all] Error 2
Erasing page at address 0x18000.
Erasing page at address 0x19000.
Erasing page at address 0x1A000.
Erasing page at address 0x1B000.
Erasing page at address 0x1C000.
Erasing page at address 0x1D000.
Erasing page at address 0x1E000.
Erasing page at address 0x1F000.
Erasing page at address 0x20000.
Erasing page at address 0x21000.
Erasing page at address 0x22000.
Erasing page at address 0x23000.
Erasing page at address 0x24000.
Erasing page at address 0x25000.
Applying system reset.
Checking that the area to write is not protected.
Programming device.
# _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
# _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
# watch_fast: "/org/gnome/terminal/legacy/" (establishing: 0, active: 0)
# unwatch_fast: "/org/gnome/terminal/legacy/" (active: 0, establishing: 1)
# watch_established: "/org/gnome/terminal/legacy/" (establishing: 0)
[100%] Built target START_JLINK
Applying system reset.
Run.
[100%] Built target FLASH_SOFTDEVICE
make: *** [Makefile:84: all] Error 2
```

Any ideas? Thanks!! :bowtie:

JF002, 2020-01-17:
There was an error in the documentation (Readme.md). When you type ```make -j```, it builds all the targets at once : the code + additional targets that programs the pinetime with JLink.

You should use this command to just build the project : ```$ make -j pinetime-app```

I've just commited an update to the doc that takes this into account.

ishotjr, 2020-01-18:
@JF002 Thanks! I built per above then flashed via:

```bash
make FLASH_SOFTDEVICE
make FLASH_pinetime-app
```

and everything works correctly, but the display is blank - is this expected? To be clear, I've had no problems w/ other projects, the display works fine, etc. Thanks! :bowtie:

JF002, 2020-01-18:
No, this is not expected... The display should show something (battery level, and the time, 00:00 by default).
Did you try to push the button to see if it wakes the display up?

ishotjr, 2020-01-19:
@JF002 yes I pushed the button. :point_right: :play_or_pause_button: 

The softdevice (and app) seems to be flashing OK:

```bash
make FLASH_SOFTDEVICE
[100%] flashing SoftDevice
Parsing hex file.
Erasing page at address 0x0.
Erasing page at address 0x1000.
Erasing page at address 0x2000.
Erasing page at address 0x3000.
Erasing page at address 0x4000.
Erasing page at address 0x5000.
Erasing page at address 0x6000.
Erasing page at address 0x7000.
Erasing page at address 0x8000.
Erasing page at address 0x9000.
Erasing page at address 0xA000.
Erasing page at address 0xB000.
Erasing page at address 0xC000.
Erasing page at address 0xD000.
Erasing page at address 0xE000.
Erasing page at address 0xF000.
Erasing page at address 0x10000.
Erasing page at address 0x11000.
Erasing page at address 0x12000.
Erasing page at address 0x13000.
Erasing page at address 0x14000.
Erasing page at address 0x15000.
Erasing page at address 0x16000.
Erasing page at address 0x17000.
Erasing page at address 0x18000.
Erasing page at address 0x19000.
Erasing page at address 0x1A000.
Erasing page at address 0x1B000.
Erasing page at address 0x1C000.
Erasing page at address 0x1D000.
Erasing page at address 0x1E000.
Erasing page at address 0x1F000.
Erasing page at address 0x20000.
Erasing page at address 0x21000.
Erasing page at address 0x22000.
Erasing page at address 0x23000.
Erasing page at address 0x24000.
Erasing page at address 0x25000.
Applying system reset.
Checking that the area to write is not protected.
Programming device.
Applying system reset.
Run.
[100%] Built target FLASH_SOFTDEVICE

make FLASH_pinetime-app
[ 98%] Built target pinetime-app
[100%] flashing pinetime-app.hex
Parsing hex file.
Erasing page at address 0x26000.
Erasing page at address 0x27000.
Erasing page at address 0x28000.
Erasing page at address 0x29000.
Erasing page at address 0x2A000.
Erasing page at address 0x2B000.
Erasing page at address 0x2C000.
Erasing page at address 0x2D000.
Erasing page at address 0x2E000.
Erasing page at address 0x2F000.
Erasing page at address 0x30000.
Erasing page at address 0x31000.
Erasing page at address 0x32000.
Erasing page at address 0x33000.
Erasing page at address 0x34000.
Erasing page at address 0x35000.
Erasing page at address 0x36000.
Erasing page at address 0x37000.
Erasing page at address 0x38000.
Erasing page at address 0x39000.
Erasing page at address 0x3A000.
Erasing page at address 0x3B000.
Erasing page at address 0x3C000.
Erasing page at address 0x3D000.
Erasing page at address 0x3E000.
Erasing page at address 0x3F000.
Erasing page at address 0x40000.
Erasing page at address 0x41000.
Erasing page at address 0x42000.
Erasing page at address 0x43000.
Erasing page at address 0x44000.
Erasing page at address 0x45000.
Erasing page at address 0x46000.
Erasing page at address 0x47000.
Erasing page at address 0x48000.
Erasing page at address 0x49000.
Erasing page at address 0x4A000.
Erasing page at address 0x4B000.
Erasing page at address 0x4C000.
Erasing page at address 0x4D000.
Erasing page at address 0x4E000.
Erasing page at address 0x4F000.
Applying system reset.
Checking that the area to write is not protected.
Programming device.
Applying system reset.
Run.
[100%] Built target FLASH_pinetime-app
```

Is there anything else you'd like me to try to help determine what might be happening? :bowtie:

JF002, 2020-01-19:
Well I've no idea :/
Could you try to update your code to the latest commit? In 8daa830ee0d33d54c0fa98bd30aadf28187d3462, a hardware reset if applied to the display, because sometimes, the display would not wakeup from sleep. Maybe it'll solve your issue?

ishotjr, 2020-01-19:
@JF002 OK, so this is really weird, but I can repro it reliably, so...LMK what you think: I pulled the latest and flashed - same blank screen as always; so I tried running `RTT terminal` to see if any output etc. (none) - and then - I tried re-flashing (while `RTT terminal` was running) - and after flashing the softdevice, the display came alive for the first time ever! Cool! So I flashed the app, and the display went blank. :face_with_head_bandage:  I then flashed the [bosmoment](https://github.com/bosmoment/PineTime-apps) firmware as a palate cleanser, and the display resulted in a mixed state of both - e.g. your date was still mostly visible (guessing there was no full refresh). So then I stopped `RTT terminal` and flashed the softdevice, then your firmware - blank as always; so I re-flashed the softdevice like last time: nothing - in fact, an error:

 ```bash
.
.
Erasing page at address 0x25000.
Applying system reset.
Checking that the area to write is not protected.
Programming device.
Segmentation fault (core dumped)
make[3]: *** [src/CMakeFiles/FLASH_SOFTDEVICE.dir/build.make:58: src/CMakeFiles/FLASH_SOFTDEVICE] Error 139
make[2]: *** [CMakeFiles/Makefile2:125: src/CMakeFiles/FLASH_SOFTDEVICE.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:132: src/CMakeFiles/FLASH_SOFTDEVICE.dir/rule] Error 2
make: *** [Makefile:131: FLASH_SOFTDEVICE] Error 2
```

Sooooo I did it again with `RTT terminal` running: SUCCESS - the display is back!! I am really confused as to why this worked, but...wanted to let you know! :dizzy_face: :rocket:

JF002, 2020-01-20:
All of this is really weird...
In my case, I only flash my firmware (haven't tried other projects yet) using jlink without any issue. Did you try to apply a (full / hard) reset on the MCU after the programming? I think it's possible using ```nrfjprog``` and ```JLinkExe```.

ishotjr, 2020-01-20:
@JF002 do you mean like `nrfjprog ---pinreset`?

JF002, 2020-01-21:
There are 3 options : ```--reset```, ```--pinreset``` and ```--debugreset```. To be honest, I don't know what are the differences between these 3 options, but you could try them out.

JF002, 2020-04-04:
@ishotjr Do you still have this issue?

JF002, 2020-09-20:
I'm closing this issue because there is no new activity for a long time...

closed 2020-09-20 by JF002