InfiniTimeOrg/InfiniTime #14: OTA

danielkucera, 2020-01-18:
I'd like to close and wear my watch. Would it be possible to implement some over-the-air update mechanism? To split it to a bootloader and main app?

JF002, 2020-01-18:
I plan on implementing such kind of feature, yes.
However, it won't be an easy task. As you said, a bootloader will be necessary.
It would be a good idea to work with the community so that all Pinetime project use the same bootloader.

Any help is welcome!

danielkucera, 2020-01-20:
It seems to be quite easy: https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/getting-started-with-nordics-secure-dfu-bootloader

JF002, 2020-01-20:
Thanks for the link!
On the chat, some people seem interested in https://mcuboot.com/

danielkucera, 2020-01-21:
I'll hopefully test this today in the evening. What chat do you mean @JF002 ?

JF002, 2020-01-21:
There are several chat rooms (Matrix, Discord, IRC, Telegram) that are linked together by bots. I'm on the Matrix one (https://riot.im/app/#/room/#pinetime64:matrix.org), but you can join from another IM app if you want. See the menu "Forum and Chat" on pine64.org.

closed 2020-06-14 by danielkucera