InfiniTimeOrg/InfiniTime #18: Help needed to use the pre-compiled binaries with BMP-bluepill and gdb

MarcusE1W, 2020-01-25:
Hi.

I was trying to use the pre-compiled binary as a first test.

I have downloaded the pinetime-app-full.hex and tried to flash it wit gdb.
Somehow that did not work. Admittedly I have not much experience with the whole subject.

Any idea what I am doing wrong?

```
(gdb) file ~/Downloads/pinetime-app-full.hex
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from ~/Downloads/pinetime-app-full.hex...(no debugging symbols found)...done.
(gdb) load
Loading section .sec1, size 0xb00 lma 0x0
Loading section .sec2, size 0xf000 lma 0x1000
Loading section .sec3, size 0x10000 lma 0x10000
Loading section .sec4, size 0x5150 lma 0x20000
Loading section .sec5, size 0xa000 lma 0x26000
Loading section .sec6, size 0x10000 lma 0x30000
Loading section .sec7, size 0x606c lma 0x40000
Start address 0x0, load size 281788
Transfer rate: 39 KB/sec, 971 bytes/write.
```

MarcusE1W, 2020-01-25:
I just read something that makes me feel uncomfortable.
Please tell me the command I was missing was not
```
run
```

?

JF002, 2020-01-26:
From your logs, flashing seems succesful, indeed. I did some tests using a black magic probe and I think I remember I had to use commands like 'run' or 'start' to run the program with the debugger, or 'kill' to disconnect the debugger and let the MCU run alone. 
The best way is to try :)

darnel, 2020-01-26:
I'm using `detach` to "finish" flashing.
https://gist.github.com/darnel/dac1370d057e176386ca4026418abc2b#flashing-pinetime

JF002, 2020-01-26:
Thanks, I'll try to replace kill by detach and see if it still works!

Did you manage to flash and run the application on your pinetime?

MarcusE1W, 2020-01-26:
I have borrowed it, I will try later.

closed 2020-01-26 by MarcusE1W

darnel, 2020-01-27:
Thanks for CMake config for gdb! After some tries I flashed using `make FLASH_SOFTDEVICE` and `make FLASH_pinetime-app` 👍 thru bluepill-bmp

JF002, 2020-01-27:
Thanks for the update! I'm happy it finally worked!