InfiniTimeOrg/InfiniTime #21: CTS is not working (for me)

darnel, 2020-01-27:
I flashed application version 0.2.1 (dff0d74) and `nRF5_SDK_15.3.0_59ac345/components/softdevice/s132/hex/s132_nrf52_6.1.1_softdevice.hex`, but I can't make current time set on PineTime. I'm using nRF Connect on iPhone to advertise CTS.
Even previous versions, manual flash or flash by `make` didn't change situation.

Another project - RIOT - works. Did I miss something obvious? Thanks.

JF002, 2020-01-27:
I've also noticed that CTS does not work as reliably as I would like, and I'll have to investigate this (i'm a newbie in BLE programming).
Note that you have to "bond" your smartphone with your pinetime, otherwise, CTS does not work at all.

darnel, 2020-01-28:
I can't find or identify pinetime device to pair either in nRF Connect or phone's BT settings. It looks like watch didn't advertises itself? However I can see "BLE" in top left of display.

darnel, 2020-02-10:
I tried using nRF Connect on Android (old 4.4 unfortunatelly), is synced but it's not correct e.g. ~now
- Android phone 10. 2. 2020 7:24
- synced time 9. 2. 2020 17:54
which is weird for me, because it doesn't look like timezone shift or something.
No way I can claim it is bug in your software, just reporting it :)

JF002, 2020-02-10:
Well, it's certainly a but in the software, and thank you for reporting it. I have to admin that I did not put a lot of effort in implementing and debugging BLE, and I (we?) will have to fix that in the future :)

Here's something you can try: in main.cpp, in function SystemTask(), you can set the variable erase_bonds to true, and run the code. Then, set it back to false and try to connect and bond. 
I think that the BLE softdevice stores the bonding information in flash, and they are erased when this variable is set.
If it works... we'll try to implement something a bit more user friendly!

JF002, 2020-02-15:
I've managed to reproduce the sync'ed time that is not correct. 
At first sight, the issue seems to be in DateTime::SetTime() and DateTime::UpdateTime().
Indeed, the logs show that the time coming from CTS is correct, but not the time converted into std::chrono::time_point. 

Here are the logs:

```
<info> app: Connected
<info> ble_cts_c: Current Time Service discovered at peer.
<info> app: CTS 0
<info> app: Current Time Service discovered on server.
<info> app: CTS 3
<info> app: Current Time received.
<info> app: 
Current Time:
<info> app: 
Date:
<info> app: 	Day of week   Saturday
<info> app: 	Day of month  15
<info> app: 	Month of year February
<info> app: 	Year          2020
<info> app: 
Time:
<info> app: 	Hours     17
<info> app: 	Minutes   21
<info> app: 	Seconds   32
<info> app: 
Adjust reason:
<info> app: 	Daylight savings 0
<info> app: 	Time zone        0
<info> app: 	External update  0
<info> app: 	Manual update    0
<info> app: 15 2 2020 
<info> app: 17 21 32 
<info> app: * 3 50 38 
<info> app: * 15 2 2020 
```

Time coming from CTS:
```
<info> app: 15 2 2020 
<info> app: 17 21 32 
```

Once it's converted in timepoint:
```
<info> app: * 3 50 38 
<info> app: * 15 2 2020 
```

darnel, 2020-02-15:
It’s great, because since your previous message I tried sync several times with either iOS or Android device and I was unable to leave 1970 :)
I changed variable to true, made app, flashed, change it back, made and flashed again...
 “BLE” text appeared and disappeared correctly according to connection (at least) to iPhone, but time didn’t change. Maybe I’m doing something wrong in whole procedure.

JF002, 2020-03-03:
I've just merged a lot of changes and this bug *should* be fixed.
@darnel Could you give version 0.3.0 a try?

darnel, 2020-03-14:
Sorry for late response, I gave a try, but I'm still stuck on not syncing anyway, and then I was on holiday... I'll try to try again :-D

darnel, 2020-05-21:
I tried 0.4.0 and today 0.5.0 but still I can't make CTS work. Pair/unpair works good with iPhone, Android phone and Debian notebook. But when I advertise CTS service from the phone, time is not synced to pinetime.

endian-albin, 2020-07-13:
> I'm using nRF Connect on iPhone to advertise CTS.

I believe this is the problem and not JF's firmware. I can also not sync current time using nRF Connect on iOS and I haven't seen anyone who can. It works on Android though, and in GNU/Linux using [this](https://github.com/bosmoment/gatt-cts) Python script.

@JF002, syncing works fine for me. I think you can close this issue.

JF002, 2020-07-13:
I've made *a lot* of improvements on the BLE connectivity, and connection and communication looks far better now than in 0.4.x or 0.5.x. Hopefuly this issue has been fixed. Please reopen this issue or open a new one if you still have issues with BLE.

closed 2020-07-13 by JF002