InfiniTimeOrg/InfiniTime #26: Compiling Against nRF52 SDK V16.x

jones139, 2020-04-30:
I am trying to build this against the nRF52 SDK V16.0.0.
There are a couple of code changes needed which are easy, and we need to include a new nrf_ble_gq library in.    I have got the compilation stage working, but the link stage is failing because it fails with errors like: 
   "ble_db_discovery.c:319: undefined reference to `nrf_ble_gq_item_add'"
Now that is defined in nrf_ble_gq/nrf_ble_gq.c, which seems to have been compiled ok:
   "Building C object src/CMakeFiles/pinetime-app.dir/home/graham/osd/OpenSeizureDetector/pineWatchSD/nRF5_SDK/componen
ts/ble/nrf_ble_gq/nrf_ble_gq.c.o"

I am not at all familiar with cmake, so I think I must have missed something trivial to get it to link the nrf_ble_gq.c.o file.   
I just wondered if you knew how to do it before I spend another evening trying to fathom it out?
My changes are here: https://github.com/OpenSeizureDetector/Pinetime

Thanks!
Graham.

JF002, 2020-05-03:
Hello,
I haven't tried to update the SDK to version 16.0.0 yet. Does it bring something useful for the project?

For the linking issue, did you add the file `nrf_ble_gq/nrf_ble_gq.c` in the ${SOURCE_FILES} variable in src/CMakeLists.txt? This specifies to cmake that it must build this .c file and link it in the final executable.

jones139, 2020-05-03:
Thanks - I don't know what extra V16.0 gives, it is just that I am setting up a new installation so was trying to use the most recent versions.   I am pretty sure I did the change to CMakeLists that you suggest, because the nrf_ble_gq.c does compile - it just doesn't seem to be linking.....but I wonder if the order in the file is important  - I might try re-ordering and see if that cures it.
Graham.

JF002, 2020-05-05:
The order in the CMake files should not be relevant, it should build all the C/C++ files and link them.
Maybe this `nrf_ble_gq` module need a `#define` in sdk_config.h ? By defautl, most of the modules from the NRF SDK are disabled by preprocessor instructions so that the code is not built if a specific variable is not defined. For example, to enable `nrf_clock` driver, you have to add 
`
#ifndef NRF_CLOCK_ENABLED
#define NRF_CLOCK_ENABLED 1
#endif
`
in the file `sdk_config.h`

jones139, 2020-05-05:
It might be something like that - I will have to check.   From the make log it seems to have compiled ok and is in the list of object files to link, so I was struggling to see what might be wrong - but if the content of the library is commented out with a  #ifdef, that could explain it.   (I have it compiling ok under 15.3.0 SDK so not too worried about this).
Thanks!

endian-albin, 2020-07-13:
The other day I could also not build using SDK v16, but I tried it just now and it worked on v 0.7.0.

endian-albin, 2020-07-13:
It works on SDK v17 as well. I've built, flashed and synchronized time and date. @JF002, I believe you can close this issue :)

JF002, 2020-07-13:
I use less and less part of the SDKm that might explain why it compiles now :)

closed 2020-07-13 by JF002