InfiniTimeOrg/InfiniTime #29: Development Plans?

jones139, 2020-05-04:
Hi, Sorry about raising an issue, but I couldn't find out where you would like to have discussions about the project.
What you have here looks like a really good start for an App that I need to develop. My application is simpler though because I don't want a generic 'end user' system, but make the watch a single use device.   I really want to use the PineTime as what I had started to design as a custom hardware epileptic seizure detector (https://www.openseizuredetector.org.uk/).   It will repeatedly collect accelerometer and heart rate data, and sent it up to the phone using BLE every 5 seconds.   The phone will respond with  a status message that will appear on the screen.
So as you seem to have BLE pretty much working, the main thing I want to do next is get the acceleromter and heart rate data - I have done this on a different microcontroller using FreeRTOS, so hoping I can port it fairly easily.
What I have in mind is to add these in a branch of your app (and maybe make mine accessible on one of the 'tiles' in your main screen for testing, but then do a different build for my deployable app where it works as a single use device.   
I wondered what you thought about that?   The other option is for me to essentially convert your work into a single use app, then add my bits, but then it would be difficult for you to make use of it.
Let me know what you would prefer.

Thanks
Graham (graham@openseizuredetector.org.uk)

JF002, 2020-05-05:
Hi!
I'll send you an email so that we can further discuss this if you want to :)

Thanks for your interest in this project. Your project is a nice real-life use case for Pinetime, and I would be happy to help you achieve it!

I have not decided on the exact license (GPL/LGPL/other, see https://github.com/JF002/Pinetime/issues/22), but I intend to keep it open-source, so, using my project as a base for yours should not be an issue.

I see 2 options for your project, depending on how you want to work:
 - Work in a 'feature' branch of this project and eventually create a pull-request to merge your work into the project. Your application could be enabled at compilation time (via preprocessor instructions `#define` for example) or at runtime via the application list, for example.
 - Fork the project, remove what you don't need and add your code.

The 1st option is my favorite as we will work together to make it possible. For example, if you write a heart-rate sensor driver, it'll be automatically available for the whole project and for other application that need the heart rate data. If I manage to make the OTA update work, you'll be able to update the application (and the whole firmware) via BLE. We can add customization if needed (like the possibility to enable/disable some features at compilation time or at runtime).

The 2nd one will allow you to better/faster customize the firmware as you'll only have to think about your application and forget about all the other features and use-cases but we won't share code as easily.

So... it's up to you ;)

jones139, 2020-05-05:
Thank you for getting back to me - I don't get hung up on Licences, I just tend to say GPL because that was the first one I came across, so it would suit me if whichever licence you pick is compatible with GPL.
I will have a go at the first option - I have a fork of your project in my OpenSeizureDetector project.   I will first work in a branch to just add the hardware access libraries so you can have access to heart rate and accelerometer data - you should then be able to just merge that back into this project if you would like it.     I will try and make one of your applets (I am not sure what the term is in the UI library you are using) that displays accelerometer and heart rate data as an example.
Once I have done that I will think about how to make my single use app version - as you say doing it with a compiler directive would be neat.
I would definitely be interested in merging your OTA work into it, because getting the app onto the watch will be the biggest hurdle.
Thanks!
Graham.

JF002, 2020-05-07:
Thanks in advance for your contributions ;)

closed 2020-05-07 by JF002