InfiniTimeOrg/InfiniTime #32: Unable to comple

danielkucera, 2020-06-14:
two separate errors here:
```
 fatal error: sdk/modules/nrfx/mdk/nrf.h: No such file or directory
```
```
file /home/danman/Projects/pinetime/Pinetime/nRF5_SDK_15.3.0_59ac345/modules/nrfx/mdk/nrf.h
/home/danman/Projects/pinetime/Pinetime/nRF5_SDK_15.3.0_59ac345/modules/nrfx/mdk/nrf.h: C source, ASCII text, with CRLF line terminators
```
and:
```
/tmp/cclA05JU.s:29: Error: selected processor does not support requested special purpose register -- `msr msp,r0'
```
full log:
```
danman@silverhorse:~/Projects/pinetime/Pinetime/build$ ls
danman@silverhorse:~/Projects/pinetime/Pinetime/build$ cmake -DNRF5_SDK_PATH=/home/danman/Projects/pinetime/Pinetime/nRF5_SDK_15.3.0_59ac345 -DUSE_OPENOCD=1 -DARM_NONE_EABI_TOOLCHAIN_PATH=/home/danman/Projects/pinetime/gcc-arm-none-eabi-8-2019-q3-update ../
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
BUILD CONFIGURATION
-------------------
    * Version : 0.7.0
    * Toolchain : /home/danman/Projects/pinetime/gcc-arm-none-eabi-8-2019-q3-update
    * NRF52 SDK : /home/danman/Projects/pinetime/Pinetime/nRF5_SDK_15.3.0_59ac345
    * Programmer/debugger : OpenOCD Client
    * Debug pins : Disabled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/danman/Projects/pinetime/Pinetime/build
danman@silverhorse:~/Projects/pinetime/Pinetime/build$ make -j pinetime-app
Scanning dependencies of target pinetime-app
[  0%] Building CXX object src/CMakeFiles/pinetime-app.dir/Logging/NrfLogger.cpp.o
[  0%] Building CXX object src/CMakeFiles/pinetime-app.dir/BlinkApp/BlinkApp.cpp.o
[  0%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/Clock.cpp.o
[  0%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/Message.cpp.o
[  2%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/Screen.cpp.o
[  2%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/DisplayApp.cpp.o
[  2%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/Tile.cpp.o
[  4%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/Gauge.cpp.o
[  4%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/Meter.cpp.o
[  4%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/Modal.cpp.o
[  4%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/BatteryIcon.cpp.o
[  4%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/BleIcon.cpp.o
[  4%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/Brightness.cpp.o
[  4%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/Label.cpp.o
[  6%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/ScreenList.cpp.o
[  6%] Building CXX object src/CMakeFiles/pinetime-app.dir/DisplayApp/Screens/FirmwareUpdate.cpp.o
[  6%] Building CXX object src/CMakeFiles/pinetime-app.dir/main.cpp.o
[  6%] Building CXX object src/CMakeFiles/pinetime-app.dir/drivers/St7789.cpp.o
[  8%] Building CXX object src/CMakeFiles/pinetime-app.dir/drivers/SpiNorFlash.cpp.o
[  8%] Building CXX object src/CMakeFiles/pinetime-app.dir/drivers/SpiMaster.cpp.o
[  8%] Building CXX object src/CMakeFiles/pinetime-app.dir/drivers/Spi.cpp.o
[  8%] Building CXX object src/CMakeFiles/pinetime-app.dir/drivers/Watchdog.cpp.o
[  8%] Building CXX object src/CMakeFiles/pinetime-app.dir/drivers/DebugPins.cpp.o
[  8%] Building CXX object src/CMakeFiles/pinetime-app.dir/drivers/InternalFlash.cpp.o
[ 11%] Building CXX object src/CMakeFiles/pinetime-app.dir/Components/Battery/BatteryController.cpp.o
[ 11%] Building CXX object src/CMakeFiles/pinetime-app.dir/Components/DateTime/DateTimeController.cpp.o
[... 211 lines cut ...]
                                               ^~~~~~~~~~~~~
/home/danman/Projects/pinetime/Pinetime/src/./Components/Ble/DeviceInformationService.h:28:45: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
         static constexpr char* fwRevision = "0.5.0";
                                             ^~~~~~~
/home/danman/Projects/pinetime/Pinetime/src/./Components/Ble/DeviceInformationService.h:29:45: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
         static constexpr char* hwRevision = "1.0.0";
                                             ^~~~~~~
In file included from /home/danman/Projects/pinetime/Pinetime/src/DisplayApp/Screens/ScreenList.h:4,
                 from /home/danman/Projects/pinetime/Pinetime/src/DisplayApp/Screens/ScreenList.cpp:3:
/home/danman/Projects/pinetime/Pinetime/src/./Components/Ble/NimbleController.h:39:45: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
         static constexpr char* deviceName = "Pinetime-JF";
                                             ^~~~~~~~~~~~~
/tmp/ccFKrlaw.s: Assembler messages:
/tmp/ccFKrlaw.s:812: Error: selected processor does not support `sev' in ARM mode
/tmp/ccFKrlaw.s:815: Error: selected processor does not support `dsb 0xF' in ARM mode
/tmp/ccFKrlaw.s:818: Error: selected processor does not support `isb 0xF' in ARM mode
/tmp/ccFKrlaw.s:1200: Error: selected processor does not support requested special purpose register -- `mrs r3,primask'
/tmp/ccFKrlaw.s:1209: Error: selected processor does not support `cpsid i' in ARM mode
/tmp/ccFKrlaw.s:1245: Error: selected processor does not support `cpsie i' in ARM mode
make[3]: *** [src/CMakeFiles/pinetime-app.dir/build.make:271: src/CMakeFiles/pinetime-app.dir/main.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:208: src/CMakeFiles/pinetime-app.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:215: src/CMakeFiles/pinetime-app.dir/rule] Error 2
make: *** [Makefile:170: pinetime-app] Error 2
danman@silverhorse:~/Projects/pinetime/Pinetime/build$ 
```

JF002, 2020-06-14:
For the 1st issue, could you edit  the first line of src/drivers/InternalFlash.cpp ? Replace
`#include <sdk/modules/nrfx/mdk/nrf.h>`
by
`#include <mdk/nrf.h>`

I couldn't reproduce the second issue using CLion. This is because CLion automatically adds -DCMAKE_BUILD_TYPE=xxx to the cmake command line.
Could you add this to your cmake command line:
`-DCMAKE_BUILD_TYPE=Release` or `-DCMAKE_BUILD_TYPE=Debug` and see if it fixes the issue on your side?

There's certainly an issue in CMakeLists.txt, because specifying CMAKE_BUILD_TYPE should not be mandatory...

Could you tell me if the build is successful with these 2 fixes?

danielkucera, 2020-06-14:
Yes, after applying both, it works

JF002, 2020-06-14:
Thx, I'll fix them soon(c) :)

danielkucera, 2020-06-14:
There is one more when I try to run it in travis:
https://travis-ci.org/github/danielkucera/Pinetime

danielkucera, 2020-06-14:
Okay, it works now, thanks.

closed 2020-06-14 by danielkucera

JF002, 2020-06-16:
Fixed in 0.6.1 - https://github.com/JF002/Pinetime/tree/0.6.1