InfiniTimeOrg/InfiniTime #35: nrf_common.ld: No such file in directory

darnel, 2020-06-17:
This error re-appeared. I know we solved it somehow, but I can't remember :-/
```
/usr/local/Caskroom/gcc-arm-embedded/9-2019-q4-major/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file ./nrf_common.ld: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/pinetime-graphics.dir/build.make:846: src/pinetime-graphics.out] Error 1
make[1]: *** [CMakeFiles/Makefile2:174: src/CMakeFiles/pinetime-graphics.dir/all] Error 2
make: *** [Makefile:104: all] Error 2
```

JF002, 2020-06-17:
I thought someone (you?) created an issue about this in the past, but I can't find it...
I think we solved it by adding the folder '../' in *target_include_directories* in src/CMakeLists.txt. 

I don't understand why this issue re appeared  and why I can't reproduce it on my setup...

Is your build folder a subfolder of the project?

darnel, 2020-06-17:
Thanks to pointing me. Using `git log | grep -B4 nrf_common.ld` I found it:
```

JF002, 2020-06-19:
Nice, thank you!
So, it fixes the issue on you side?

Could you change the destination branche from master to develop? 
I merge into master only when I want to generate a release, whereas develop is the current 'main' branch.

Thanks!

darnel, 2020-06-19:
Yes it fixed the issue for me. Destination changed.

JF002, 2020-06-20:
Thanks, merged into develop :)

closed 2020-06-20 by JF002