InfiniTimeOrg/InfiniTime #48: Add github actions CI badge to readme.md

incyi, 2020-07-28:
It is possible to show the build status like this: [image]

I can not make a pull request now, because I have a lot of changes.

Avamander, 2020-07-28:
You can use git's "keep" reset for quick commits. Just commit the specific changes that add the badge, push those changes to a `patch-{n}` branch, then do a keep reset back to the master. If everything went well, now that specific change is gone sitting on a branch and you can work on the rest of your code. I really do recommend testing it out and learning how to use it, it's very nice.

Using soft reset or hard reset would not be as nice, the former would leave all changes that differ from master and the latter would wipe local changes that differ from master.

incyi, 2020-12-09:
Fixed

closed 2020-12-09 by incyi