InfiniTimeOrg/InfiniTime #52: Build issue

thibww67, 2020-08-11:
I have issues to build the project, I download the cross-compiler, the NRF52 SDK, and MinGW.

My command line is : 

`cmake -DARM_NONE_EABI_TOOLCHAIN_PATH="c:/program files (x86)/gnu tools arm embedded/8 2019-q3-update/" -DNRF5_SDK_PATH="C:\Projects\Pinetime\sdk" -DCMAKE_BUILD_TYPE=DEBUG -DUSE_JLINK=1 -DNRFJPROG="c:/program files (x86)/segger/jlink" -G "MinGW Makefiles" -DCMAKE_C_COMPILER="c:/program files (x86)/gnu tools arm embedded/8 2019-q3-update/bin/arm-none-eabi-gcc.exe" -DCMAKE_CXX_COMPILER="c:/program files (x86)/gnu tools arm embedded/8 2019-q3-update/bin/arm-none-eabi-g++.exe" -DCMAKE_ASM_COMPILER="c:/program files (x86)/gnu tools arm embedded/8 2019-q3-update/bin/arm-none-eabi-as.exe" ../`

And the output on my Windows 10 is : 

-- The C compiler identification is GNU 8.3.1
-- The CXX compiler identification is GNU 8.3.1
-- The ASM compiler identification is GNU
-- Found assembler: c:/program files (x86)/gnu tools arm embedded/8 2019-q3-update/bin/arm-none-eabi-as.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: c:/program files (x86)/gnu tools arm embedded/8 2019-q3-update/bin/arm-none-eabi-gcc.exe
-- Check for working C compiler: c:/program files (x86)/gnu tools arm embedded/8 2019-q3-update/bin/arm-none-eabi-gcc.exe - broken
CMake Error at C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "c:/program files (x86)/gnu tools arm embedded/8 2019-q3-update/bin/arm-none-eabi-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Projects/Pinetime/build/CMakeFiles/CMakeTmp

    Run Build Command(s):C:/MinGW/bin/mingw32-make.exe cmTC_52d45/fast && C:/MinGW/bin/mingw32-make.exe  -f CMakeFiles\cmTC_52d45.dir\build.make CMakeFiles/cmTC_52d45.dir/build
    mingw32-make.exe[1]: Entering directory 'C:/Projects/Pinetime/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_52d45.dir/testCCompiler.c.obj
    c:\PROGRA~2\GNUTOO~1\82019-~1\bin\AR19DD~1.EXE    -o CMakeFiles\cmTC_52d45.dir\testCCompiler.c.obj -c C:\Projects\Pinetime\build\CMakeFiles\CMakeTmp\testCCompiler.c
    Linking C executable cmTC_52d45.exe
    "C:\Program Files\CMake\bin\cmake.exe" -E cmake_link_script CMakeFiles\cmTC_52d45.dir\link.txt --verbose=1
    "C:\Program Files\CMake\bin\cmake.exe" -E rm -f CMakeFiles\cmTC_52d45.dir/objects.a
    C:\PROGRA~2\GNUTOO~1\82019-~1\bin\ARM-NO~2.EXE cr CMakeFiles\cmTC_52d45.dir/objects.a @CMakeFiles\cmTC_52d45.dir\objects1.rsp
    c:\PROGRA~2\GNUTOO~1\82019-~1\bin\AR19DD~1.EXE -Wl,--whole-archive CMakeFiles\cmTC_52d45.dir/objects.a -Wl,--no-whole-archive -o cmTC_52d45.exe -Wl,--out-implib,libcmTC_52d45.dll.a -Wl,--major-image-version,0,--minor-image-version,0 
    c:/progra~2/gnutoo~1/82019-~1/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld.exe: unrecognized option '--major-image-version'
    c:/progra~2/gnutoo~1/82019-~1/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld.exe: use the --help option for usage information
    collect2.exe: error: ld returned 1 exit status
    CMakeFiles\cmTC_52d45.dir\build.make:105: recipe for target 'cmTC_52d45.exe' failed
    mingw32-make.exe[1]: *** [cmTC_52d45.exe] Error 1
    mingw32-make.exe[1]: Leaving directory 'C:/Projects/Pinetime/build/CMakeFiles/CMakeTmp'
    Makefile:139: recipe for target 'cmTC_52d45/fast' failed
    mingw32-make.exe: *** [cmTC_52d45/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Projects/Pinetime/build/CMakeFiles/CMakeOutput.log".
See also "C:/Projects/Pinetime/build/CMakeFiles/CMakeError.log".

Do you see any solution ? I checked all the paths so this is not the issue.

Thank you !

lupyuen, 2020-08-11:
Hi: Building the firmware on Windows may be tricky... Would you like to try building in the GitHub Cloud with GitHub Actions? Here are the steps...

https://lupyuen.github.io/pinetime-rust-mynewt/articles/cloud

thibww67, 2020-08-11:
Thank you for your answer, I start to follow the guide, but when I build  it I get these errors : 

[image]

I modified the code as it is shown in the guide and relaunched but it is the same.

lupyuen, 2020-08-11:
Oh sorry if you browse the article a little more you'll see this fix, please try this...

If we see this error...

```
/home/runner/work/Pinetime/Pinetime/src/drivers/TwiMaster.cpp:1:10: fatal error: sdk/integration/nrfx/nrfx_log.h: No such file or directory
 #include <sdk/integration/nrfx/nrfx_log.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [src/CMakeFiles/pinetime-app.dir/drivers/TwiMaster.cpp.o] Error 1
```

Browse to...

```
https://github.com/ACCOUNT_NAME/Pinetime/blob/master/src/drivers/TwiMaster.cpp
```

(Change `ACCOUNT_NAME` to our GitHub Account Name)

Edit the first two lines...

```c
#include <sdk/integration/nrfx/nrfx_log.h>
#include <sdk/modules/nrfx/hal/nrf_gpio.h>
```

To...

```c
#include <nrfx_log.h>
#include <nrf_gpio.h>
```

Click `Commit Changes` to save the file.

This triggers a new Firmware Build, which should succeed now.

thibww67, 2020-08-11:
Thank you for your answers !

Yes I noticed this just after I wrote my comment, it works and I managed to download the ZIP file.
The last problem is that Nrf Connect seems to have change a little bit and I can't find the way to download the zip with the DFU. on Android.

[image]

lupyuen, 2020-08-11:
Cool great! I think you need to flash the wired way the first time, after that nRF Connect will work.

Look in the "Other Options" section questions 4 and 5...

https://lupyuen.github.io/pinetime-rust-mynewt/articles/cloud#other-options

Do you have a Raspberry Pi? We will need to unprotect the Flash ROM with pinetime-updater as explained here...

https://github.com/lupyuen/pinetime-updater/blob/master/README.md#remove-flash-rom-protection

Then we can use pinetime-updater to flash the file pinetime-app.out (note that it's not the same as the other ZIP file)...

1.    Download the artifact `pinetime-app.out` from GitHub Actions.

1.    We'll get a ZIP file. Extract the PineTime Firmware Image inside: `pinetime-app.out`

1.    Flash with [__PineTime Updater__](https://github.com/lupyuen/pinetime-updater/blob/master/README.md)...

    -  Select `Downloaded File` and `pinetime-app.out`

    -  Flash to address `0x0`

lupyuen, 2020-08-11:
If you're still stuck, come chat with us on the PineTime Chatroom on Discord / Matrix / Telegram / IRC: https://wiki.pine64.org/index.php/PineTime#Community

thibww67, 2020-08-11:
Yes I still stuck because I have no Raspberry available right now.. 
I just managed to launch the Cmake command with OpenOCD but it changes nothing.
I posted a thread on Pine64.
Thank you for your time !

lupyuen, 2020-08-11:
Sure no problem :-) I'm sorry that you had to get the first batch of PineTime watches, the upcoming batch will have built-in wireless flashing support.

Now that you have figured out how to build the firmware in the cloud, maybe you could experiment with Clock.cpp and make your own watch face :-)

https://lupyuen.github.io/pinetime-rust-mynewt/articles/pinetime#programming-with-pinetime

thibww67, 2020-08-11:
The project is very interesting but unfortunately I can not yet build the code on my Windows, and even if I build it, I don't really know if it is possible to flash it over USB.

But thank you for all your answers !

lupyuen, 2020-08-11:
Sure I understand. We have a newcomer who developed their own Watch Face without a PineTime, by simply editing Clock.cpp and building with GitHub Actions.

We'll have a tutorial on that soon (written by our newcomer). Check with me on Twitter or Mastodon for updates :-)

https://twitter.com/MisterTechBlog

https://qoto.org/@lupyuen

borgmanJeremy, 2020-08-11:
@thibww67 I have not tried it myself, but I think you should be able to build it on windows using windows subsystem for linux (WSL).

JF002, 2020-08-11:
I did not try to compile this project on windows, but this would be a nice addition. PR welcome :)

The [documentation](https://github.com/JF002/Pinetime/blob/develop/doc/CompanionApps/NrfconnectOTA.md) about NRFConnect is valid only if you've already programmed this project on your Pinetime. The stock firmware does not work that way.

You'll need a [SWD debugger](https://wiki.pine64.org/index.php/Reprogramming_the_PineTime) to program the firmware into your pinetime. And, as the Pinetime comes with the memory protected, you need a debugger than is able to unlock it (jlink, raspberrypi,...). 
The memory won't be locked in the next batch, but it's not available yet...

JF002, 2020-09-20:
I'm closing this issue as there's no activity for more than a month. @thibww67 reopen it if necessary.

closed 2020-09-20 by JF002