InfiniTimeOrg/InfiniTime #55: Build error in Debian

endian-albin, 2020-08-31:
After fetching the latest revision I get the this build error in Debian Bullseye:

```
/home/albin/projekt/pt/pinetime-freertos/src/drivers/TwiMaster.cpp:1:10: fatal error: sdk/integration/nrfx/nrfx_log.h: No such file or directory
 #include <sdk/integration/nrfx/nrfx_log.h>
```

I've tried various configurations, e.g. this one:

`cmake -DCMAKE_BUILD_TYPE=Debug -DARM_NONE_EABI_TOOLCHAIN_PATH=/usr/ -DNRF5_SDK_PATH=~/bin/nrf52-sdk-15.3.0/ -DUSE_JLINK=1 -DNRFJPROG=~/nrfjprog ../`

(But the way, `/usr/` is weird way of referencing `/usr/bin/`.)

lupyuen, 2020-08-31:
Try this...

If we see this error...

```
/home/runner/work/Pinetime/Pinetime/src/drivers/TwiMaster.cpp:1:10: fatal error: sdk/integration/nrfx/nrfx_log.h: No such file or directory
 #include <sdk/integration/nrfx/nrfx_log.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [src/CMakeFiles/pinetime-app.dir/drivers/TwiMaster.cpp.o] Error 1
```

Browse to...

```
https://github.com/ACCOUNT_NAME/Pinetime/blob/master/src/drivers/TwiMaster.cpp
```

(Change `ACCOUNT_NAME` to our GitHub Account Name)

Edit the first two lines...

```c
#include <sdk/integration/nrfx/nrfx_log.h>
#include <sdk/modules/nrfx/hal/nrf_gpio.h>
```

To...

```c
#include <nrfx_log.h>
#include <nrf_gpio.h>
```

From https://lupyuen.github.io/pinetime-rust-mynewt/articles/cloud#our-first-pinetime-firmware-build

endian-albin, 2020-08-31:
@lupyuen: thank you, it works! So is there a reason why this patch has not already been applied?

lupyuen, 2020-08-31:
@JF002 : Remember this? :-)

JF002, 2020-08-31:
Yes, these errors are a bit annoying. The fix is already applied on the develop branch but it's not merged yet in the master branch.

In fact, I use the branch develop as an 'integration' branch : all developments that are finished and working are merged into develop. Then, when we decide to generate a new release (with a GIT tag and binary files published on github), I merge this branch develop into master.

This is explained in this doc : https://github.com/JF002/Pinetime/blob/develop/doc/branches.md.

I think I'll release this version 0.8.0 soon, so that people can update their futur devkit with this new version.

endian-albin, 2020-08-31:
Ok! So I'll just build from the dev branch in the future then :)

endian-albin, 2020-09-11:
I've now successfully built and tested both master and v0.8.1 locally using the `gcc-arm-none-eabi-9-2019-q4-major` which worked fine.

closed 2020-09-11 by endian-albin