InfiniTimeOrg/InfiniTime #56: Cannot run mcuboot app

endian-albin, 2020-08-31:
I cannot run the bootloader-compatible version. It gets stuck at the bootloader screen.

```
$ make pinetime-mcuboot-app
$ imgtool.py create --align 4 --version 1.0.0 --header-size 32 --slot-size 475136 --pad-header src/pinetime-mcuboot-app.bin infinitime-image.bin
$ pyocd flash -e sector -a 0x8000 -t nrf52 infinitime-image.bin
```

The bootloader is lupyuen's prebuilt `mynewt_nosemi.elf.bin` (and it can boot Hypnos).

lupyuen, 2020-08-31:
Can you send me infinitime-image.bin? Lemme peek inside

endian-albin, 2020-08-31:
You can download it [here](https://cloud.fripost.org/s/zqaPtRntg8StmKT), thanks.

lupyuen, 2020-08-31:
Looks like a valid MCUBoot Header. Wonder what's wrong...

[image]

JF002, 2020-08-31:
I've tried your image file, and it doesn't work on my Pinetime either...

Can you tell me from which branch/commit you built the image? Maybe also the version of the toolchain and the cmake command line?

I've just built and flash the last commit from develop (754918fda6a1ead2c6a8b440af489aae038918d0) and it works perfectly...

endian-albin, 2020-08-31:
I built it from the master branch: "Add InfiniTime logo" (be05997272b7b1d1b25b122c8162ac6f4c1c12a2).

> does pyocd provide you other erase options (erase the whole flash, for example) ?

I'm not sure. I used `nrfjprog --ereaseall` for erasing the entire internal flash.

I just tried to build from the same commit as you on the develop branch but got several pages of errors, e.g this:

```
usr/include/newlib/c++/8.3.1/bits/basic_string.tcc:1239:3: error: template with C linkage
   template<typename _CharT, typename _Traits, typename _Alloc>
   ^~~~~~~~
/usr/include/newlib/c++/8.3.1/bits/basic_string.tcc:1261:3: error: template with C linkage
```

Build steps:
```
$ rm -rf build/
$ mkdir build/ 
$ cd build/
$ cmake -DCMAKE_BUILD_TYPE=Debug -DARM_NONE_EABI_TOOLCHAIN_PATH=/usr/ -DNRF5_SDK_PATH=~/bin/nrf52-sdk-15.3.0/ -DUSE_JLINK=1 -DNRFJPROG=~/nrfjprog ../
$ cd ..
$ make pinetime-mcuboot-app
```

endian-albin, 2020-09-11:
I've now successfully built using the  `gcc-arm-none-eabi-9-2019-q4-major` toolchain.

closed 2020-09-11 by endian-albin