InfiniTimeOrg/InfiniTime #58: Firmware Version in nRF Connect

lupyuen, 2020-09-01:
0.8.0 Release Candidate works great via DFU great job! :-) Just a minor cosmetic issue: The Firmware Revision reported by nRF Connect still shows 0.7.0 instead of 0.8.0. Thanks!

[image]

JF002, 2020-09-02:
You're the eagle eye, @lupyuen. The version reported on BLE is hard-coded (and I forgot to update it) and it should be updated automatically when the project version is changed. Should be easy to fix!

JF002, 2020-09-02:
Fixed in https://github.com/JF002/Pinetime/commit/1aa5b0a440de3015b5b4c88aa32d149a69bfa707.

closed 2020-09-02 by JF002