InfiniTimeOrg/InfiniTime #62: P8 support

nextfullstorm, 2020-09-14:
Hi,
can anyone tell me how to build Infinetime for Colmi P8? It's wide-known clone of Pinetime (or Pinetime clone of P8? I'm not sure). [This](https://github.com/atc1441/ATCwatch) awesome project proves that there is little to no difference in hardware and pinout differs. I don't have enough skill to resolve it myself. After blind dfu upgrade from stock fw to latest Infinitime release my device was bricked and I have to restore it via SWD programmer.

JF002, 2020-09-14:
Hi!
I'm not working on a P8 port for now, because I do not have one and I doubt I would have the time to maintain both of them.
You're right, there are very few differences between the PineTime and the P8, so the port should not be too complicated. I'm open to pull requests if someone wants to implement the support for the P8 :)

Please note that the DFU upgrade from ATCWatch and InfiniTime are **not** compatible. ATCWatch and InfiniTime do not use the same bootloader and memory map, which means that you can DFU from ATCWatch to InfiniTime or from InfiniTime to ATCWatch.
There is a project to build a tool that would allow that but I don't think it's ready to be used for now.

maidenone, 2020-09-14:
i have worked with DFU and NRF52, it should be possible to create a new DFU package that contains the correct bootloader.

hrmckay, 2020-09-15:
If you look at issue # 69 for the wasp-os support for the Pinetime at https://github.com/daniel-thompson/wasp-os/issues you will see what had to be changed to get things working for that operating system.

Hugh

nextfullstorm, 2020-09-15:
Tnx guys. As I understand, main problem here is memory map difference. And THAN maybe pinout/accelerometer difference. I will look into this and write back if I have meaningful result.

maidenone, 2020-09-15:
where can i buy a p8?

nextfullstorm, 2020-09-15:
> 
> where can i buy a p8?

On aliexpress for sure. I got mine there. IDK if I can post direct links.

maidenone, 2020-09-15:
well i ordered 2, will se when i get them. 
do not know when i will get the time, but i intend to look into it.

JF002, 2020-09-15:
> i have worked with DFU and NRF52, it should be possible to create a new DFU package that contains the correct bootloader.

@maidenone The tool I was talking about is [wasp-reloader](https://github.com/daniel-thompson/wasp-reloader). You might want to have a look at it before you start something from scratch :)
And thanks in advance for your futur PR for the support of the P8 in InfiniTime :)

> Tnx guys. As I understand, main problem here is memory map difference. And THAN maybe pinout/accelerometer difference. I will look into this and write back if I have meaningful result.

@nextfullstorm It's not only a memory map issue: the bootloader philosophy is completely different. ATCWatch is based on the NRF softdevice (BLE stack) and NRF bootloader. The BLE stack is shared accross the bootloader and the application (atcwatch). It provides many advantages, but it forces in a way all application firmware to use the NRF Softdevice as BLE stack. And the softdevice and the bootloader are both closed-source.

The bootloader implemented by InfiniTime and other compatible firmwares is a much more simple bootloader that does not provide any BLE stack. It allows the developers to choose the BLE stack they want for their project (or no ble at all). It is then possible to implement a fully open-source application: the bootloader is based on MCUBoot and you can use NimBLE as ble stack. They are both open source.

The way the NRFBootloader and softdevice are designed make the cohabition really difficult. The best solution for this issue we've found so far is to implement a "swticher" tool that would use OTA to overwrite the bootloader from MCUboot to NRF and vice-versa. Danielt began the project (wasp-reloader) but I haven't had the time to check it out.

nextfullstorm, 2020-09-16:
@JF002 Thanks again. This is very useful info for noobs like me. Maybe you should include it somehow in main README.

tt1pjm, 2020-09-18:
I created a fork and have flashed pinetime-app.out to a P8 that seems to be functional. Only a few pins needed to be modified. I've not yet managed to get MCUBoot bootloader working yet for the P8. If anyone makes any progress on that please share.

endian-albin, 2020-09-22:
@tt1pjm: PineTime [Hypnos](https://github.com/endian-albin/pinetime-hypnos) has a P8 port with mcuboot enabled, but I don't know if DFU works it since I don't have access to a P8 device. Please note that the slot1 partition is located on the external SPI flash. I have no idea if that is right or wrong on the P8. If you put slot1 on the the internal flash, then the Hypnos application will run out of space. Anyway, binaries are available in the Github Actions history.

tt1pjm, 2020-09-22:
@endian-albin: Good work on building support for P8 in Hypnos! I managed to get MCUboot working on my P8, take a look at my fork of Lup's pinetime-rust-mynewt for details. I don't know if all P8s have the same jedec manufacturer for SPIflash as mine but it was different to Pinetime and needed correcting for MCUBoot to work. Looks as though you've got the same jedec manufacturer for P8 as Pinetime in Hypnos so may need updating if other P8 owners validate they have the same jedec manufacturer that I've identified.

endian-albin, 2020-09-22:
The port was contributed by @MirkoCovizzi and as I said, I haven't been able to test it myself. From what you say, it sounds like I need to make modifications, but I'll leave that to someone who can actually try it out. Do you happen to know where one can get a P8 with SWD pin access?

tt1pjm, 2020-09-22:
I'll try and test the change and let you know how it goes over at Hypnos. P8s come closed and just need back cover removing with a blade to access the SWD pins. They're quite common and can be found searching for Colmi P8.

endian-albin, 2020-09-22:
Thanks @tt1pjm!

maidenone, 2020-09-26:
@endian-albin 
see my posts here: https://github.com/JF002/Pinetime/issues/53

i will set up integration testing nodes for P8 and Pinetime. they will be available online through a web interface. i can also provide direct ssh access if asked ^_^

endian-albin, 2020-09-26:
@maidenone, that sounds awesome. I'm looking forward to try it out when ready.

JF002, 2020-09-27:
Thanks for working on a port of InfiniTime on the P8. 
To avoid confusion, we'll need to be very clear about the fact that the bootloader and OTA from InfiniTime are not compatible with the original FW running on the P8 and with Wasp-OS. I don't want people to brick their device while trying InfiniTime :)

jones139, 2020-09-27:
I have just ordered a second P8 in anticipation of having to open up the first when I brick it 😀.

I'll give it a try tonight.

maidenone, 2020-10-04:
Do you have any good linux commandline tools to do make DFU upgrade requests? 

I would like to create "fw upgrade test" to my integration test framework, 
see: https://forum.pine64.org/showthread.php?tid=11711

Avamander, 2020-10-04:
> Do you have any good linux commandline tools to do make DFU upgrade requests?

@maidenone use adafruit-nrfutil pip package

maidenone, 2020-10-05:
this looks right: https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/nordicsemi/dfu/dfu_transport_ble.py

Avamander, 2020-10-05:
@maidenone 
> looks like the adafruit lib only supports serial dfu, not ble dfu.

You are absolutely correct. When I replied to you it was getting pretty late and I thought you just want to make "DFU distribution packets". For which that library works well. I used it to quickly generate the .zip needed for Gadgetbridge or nRF Connect.

maidenone, 2020-10-05:
just to clarify my plan:
The test will
1. flash bootloader + firmware/app or only bootloader if supported
2. the test framework will then try to make the bootloader enter dfu mode, either through pi hardware (button press?) or through "buttonless dfu" if available.
3. the test framework will then make a ble dfu upgrade request through RPI BLE (or NRF BLE dongle).
4. the test framework will then wait / reboot the pinetime and verify that the update succeeded (RTT output? debug trace? pin toggle?)

this will make it faster / easier to test / develop solutions for moving from one bootloader to another, verify normal dfu upgrade packages or creating tests that repeatedly can cause some hard to trigger upgrade bugs.

Avamander, 2020-10-05:
What you describe sounds very similar with what I thought of doing, but the primary difference with yours being that I don't really wish to write a custom UI/service, the status and everything will hopefully be available as build artifacts or in CI log. I was personally planning on creating a CI pipeline for TeamCity that flashes the firmware, grabs the ARM Semihosting logs, takes a few pics with a webcam and then posts the build status.

maidenone, 2020-10-05:
shhh, ARM Semihosting, isnt that a bad word?
I have had BLE timing issues with the softdevice when using semihosting
[image]

Avamander, 2020-10-05:
> shhh, ARM Semihosting, isnt that a bad word?

It works if you don't use it in timing-critical situations, plus it's not proprietary and nicely supported by OpenOCD.

maidenone, 2020-10-05:
if you want to dig deeper see this: https://blog.japaric.io/itm/
summery: "Never, ever use semihosting to log data. Unless you can afford delays in the order of tens of milliseconds in your application."

It was with strong dislike that i went from the light fields of OpenOCD into the dark plains of JlinkRTT, now i am afraid i will never see the light again... 

that said, i do a fair bit of time critical things, and being able to dump out a metric crap ton of debug data without missing realtime constraints is a true blessing.

maidenone, 2020-10-05:
> What you describe sounds very similar with what I thought of doing, but the primary difference with yours being that I don't really wish to write a custom UI/service.

you might be able to use the commandline interface?

`python3 runOnTestNode.py dfuUpgradeTest node1 test.hex`

The drawback is that i currently do not have any "job id", only "execution datetime", and if the node is busy and your job is queued, you will have a hard time reading out the test result. unless you want to hang around reading real time debug output during the entire execution.

Avamander, 2020-10-05:
> if you want to dig deeper see this: https://blog.japaric.io/itm/
> summery: "Never, ever use semihosting to log data. Unless you can afford delays in the order of tens of milliseconds in your application."

I think the author might've made the exact same mistake of not increasing the adapter speed...

endian-albin, 2020-10-05:
@maidenone, if you want to support the Hypnos project as well, new over-the-air firmware images must to be sent over the Simple Management Protocol. I use the [mcumgr](https://github.com/apache/mynewt-mcumgr#command-line-tool) command-line tool and run the following commands as root:
```
# mcumgr --conntype="ble" --connstring ctlr_name=hci0,peer_name='Hypnos' image list
# mcumgr --conntype="ble" --connstring ctlr_name=hci0,peer_name='Hypnos' image upload hypnos-mcuboot-app-img.bin
# mcumgr --conntype="ble" --connstring ctlr_name=hci0,peer_name='Hypnos' image test <hash of slot-1 image>
# mcumgr --conntype="ble" --connstring ctlr_name=hci0,peer_name='Hypnos' reset
# mcumgr --conntype="ble" --connstring ctlr_name=hci0,peer_name='Hypnos' image confirm
```
I do  provide Nordic DFU packages for InifiniTime support, but I cannot use them myself for upgrades.

Also, the only debug log available for Hypnos at the moment is JlinkRTT which is gets enabled when building with `LOGGING="on"` but not in the default "production" firmware.

maidenone, 2020-10-05:
> @maidenone, if you want to support the Hypnos project as well
Of cause i want to support all the projects i can, I will allow contributions for test files as well, but this is a big security risk, and i have put any time into creating an isolated environment... so i will simply invite developers to the test file git repo (seperated from the framework git repo). 

Once i get it up and running, creating custom test files for other projects should be very simple.

> RTT is not in the default "production" firmware.
This is as it should, hence my comment in my forum post on "requiring test debug" has its drawbacks.

kung1388, 2021-01-27:
I change following files for P8:
SystemTask.h
   line 85:
   static constexpr uint8_t pinButton = 17;		// pinetime 13
Cst816s.h
   line 42:
   static constexpr uint8_t pinReset = 13;			// pinetime 10
BatteryController.h
  line 51:
  static constexpr uint32_t chargingPin = 19;		// pinetime 12

ildar, 2021-02-02:
@kung1388 , did you test it? what exact model string you have? thanks!

ildar, 2021-02-28:
Allright,
with quick&dirty patch,here are the bootloader and the FW:
https://github.com/ildar/pinetime-mcuboot-bootloader ,
https://github.com/ildar/InfiniTime
Would anyone want to test it on *uncovered* P8?

ildar, 2021-02-28:
[InfiniTime-P8.zip](https://github.com/JF002/InfiniTime/files/6056750/InfiniTime-P8.zip)

tt1pjm, 2021-02-28:
Hi Ildar,

I've already got a working version of both bootloader and infinitime running on a P8. You'll need to make a small change to bootloader for different SPI Flash on the P8. Changes I made to bootloader are:
https://github.com/tt1pjm/pinetime-rust-mynewt/commit/621da481260ca020d38c38257106f26ab2da52b6
https://github.com/tt1pjm/pinetime-rust-mynewt/commit/3fb35bceeb40766d28229e82ed6757af46e706ee
Inifinitime works as expected with your changes.

ildar, 2021-02-28:
@tt1pjm , thanks.
I re-applied your patches and re-uploaded ZIP with BIN FW files.
Anybody, please test, or @tt1pjm , please share your binaries including DFU
zip to upload through BLE.

kung1388, 2021-03-01:
@ildar sorry, too late reply you.
I try to install InfiniTime-P8.zip. it cannot work, every time reboot.
When I download new version from JF002. I will change some PIN number with following command. Finally It is work.

fileName="./InfiniTime-0.14.0/Pinetime"

perl -pi.bak -e 's/pinButton = 13/pinButton = 17/g' $fileName/src/systemtask/SystemTask.h
perl -pi.bak -e 's/pinReset = 10/pinReset = 13/g' $fileName/src/drivers/Cst816s.h
perl -pi.bak -e 's/chargingPin = 12/chargingPin = 19/g' $fileName/src/components/battery/BatteryController.h
cd $fileName
mkdir build
cd build
cmake ../ -DCMAKE_BUILD_TYPE=Release -DARM_NONE_EABI_TOOLCHAIN_PATH=/Applications/ARM -DNRF5_SDK_PATH=~/nRF5_SDK_15.3.0 -DNRFJPROG=~/nrfjprog/9.4.0
make

kung1388, 2021-03-01:
I try to make your source code again. It's work.
I don't know why your InfiniTime-P8.zip cannot work for me. 
I program "pinetime-app-0.14.0.bin" with swd

kung1388, 2021-03-01:
@ildar, made from your source code.
[InfiniTime-0.14.0-P8.zip](https://github.com/JF002/InfiniTime/files/6061014/InfiniTime-0.14.0-P8.zip)

kung1388, 2021-03-01:
> @kung1388 , did you test it? what exact model string you have? thanks!

[image]

ildar, 2021-03-01:
@kung1388 , thanks for testing.
1. That ZIP file isn't DFU OTA. It is just the pack of FW files for
flashing through SWD.
2. It's good that "pinetime-app-0.14.0.bin" works.
3. Now could you please flash the bootloader (mynewt.elf.bin) to 0x0000 and
infinitime FW (pinetime-mcuboot-app-0.14.0.bin) to 0x8000? (details here:
https://github.com/JF002/InfiniTime/tree/develop/bootloader#bootloader-binary
)
thanks

kung1388, 2021-03-01:
I try to build image.bin has problem. I do it first.
imgtool create --align 4 --version 1.0.0 --header-size 32 --slot-size 475136 --pad-header ./pinetime-mcuboot-app-0.14.0.bin image.bin

kung1388, 2021-03-01:
@ ildar, 
[dfu.zip](https://github.com/JF002/InfiniTime/files/6061807/dfu.zip)
work fine with OTA. But this dfu made by genpkg. 
I want to change it with MAC address but not success.

ildar, 2021-03-01:
Try this: [InfiniTime-P8.zip](https://github.com/JF002/InfiniTime/files/6063051/InfiniTime-P8.zip)
It is the combined bootloader + Infinitime. Just flash it to 0x000

kung1388, 2021-03-01:
It's work fine.

kung1388, 2021-03-01:
I cannot make pinetime-mcuboot-app-0.14.0.bin and mynewt.elf.bin into one file.
How to do?

ildar, 2021-03-01:
On Mon, Mar 1, 2021 at 10:58 PM kung1388 wrote:

> I cannot make pinetime-mcuboot-app-0.14.0.bin and mynewt.elf.bin into one
> file.
> How to do?
>
The https://github.com/JF002/InfiniTime/issues/62#issuecomment-788093612 is
exactly that

Ideally you should see this:
https://video.codingfield.com/videos/watch/5b70cc41-2e14-49cc-a631-0aa466957169
But I guess you haven't run "graphics" ROM so I don't know what you see
during the bootloader phase.

kung1388, 2021-03-02:
I modified the Pinetime code for P8.
https://github.com/kung1388/InfiniTime-P8.git
and
https://github.com/kung1388/mcuboot-bootloader-P8.git

ildar, 2021-03-06:
There is a way to install InfiniTime OTA through BLE. It is
`reloader-factory.zip`. It should mainly be installed using wasp-os
bootloader but might be possible over any other standard Nordic DFU
bootloader.
Please test this one reloader-factory.zip (ONLY IF YOU HAVE SWD to fix
possible error).
The source goes here: https://github.com/ildar/wasp-reloader

JF002, 2021-03-06:
Wow, nice work on this port of InfiniTime on the P8!

I see there's some interests on a version of InfiniTime on this device. I don't plan on doing and maintaining it myself (I don't have the hardware, and the time to do it), but if someone wants to do a full port (InfiniTime, bootloader, documentation, test, maintenance,...), we can try to find a way to integrate it nicely in the codebase.

ildar, 2021-03-09:
On Sat, Mar 6, 2021 at 5:00 PM Ildar wrote:

> Please test this one reloader-factory.zip (ONLY IF YOU HAVE SWD to fix
> possible error).
>

Also the DFU file to get back to WASP-OS attached.

ildar, 2021-03-11:
Another thing to try:
Having a SoftDevice bootloader (e.g. Espruino or ATCwatch or WASP-OS
bootloader), this DFU package would switch you to the pinetime-bootloader
and install recovery. Then InfiniTime should be installed from the recovery
following normal procedure.
Needs testing on a P8 watch!

ildar, 2021-03-13:
@tt1pjm , on
https://github.com/JF002/InfiniTime/issues/62#issuecomment-696742639, I
opened separate issue:
https://github.com/JF002/pinetime-mcuboot-bootloader/issues/6

ildar, 2021-03-13:
Status update:
Currently working:
1. Infinitime installed without bootloader
2. Bootloader (2 FW files for different SPI flash found in different P8)
3. pinetime-mcuboot-recovery-loader-image-0.14.1.bin running from the BL.
Not working:
1. all other FW aren't accepted by BL, causing bootloop
2. recovery, even though installed by loader, doesn't start by BL with
button press

I guess @tt1pjm has better results, I'd want to know details.

morsedev, 2021-03-18:
Woow, great work!!, What is the status of this research now guys??

ildar, 2021-03-19:
Sorry, port isn't yet complete.
1. standalone InfiniTime flashed with SWD works fine. I use it daily.
2. bootloader starts and sometimes launches some images (see above). But
not (yet) the InfiniTime.
3. Recovery also doesn't start from the BL.

ildar, 2021-03-19:
latest builds
[InfiniTime.zip](https://github.com/JF002/InfiniTime/files/6169074/InfiniTime.zip)

ildar, 2021-03-31:
Allright.
Just published the final version of InfiniTime fw for P8 watch. Files are in the post above ↑
1. InfiniTime+BL-0.15.0-P8.hex : Bootloader+InfiniTime, for flashing of opened watch (like [this](https://wiki.pine64.org/wiki/PineTime_Devkit_Wiring#SWD_Pinout)
2. IT_recovery-loader+BL-0.15.0.hex : Bootloader + recovery installer, for flashing of opened watch
3. reloader-factory.zip is for OTA installation from SoftDevice-based bootloaders (like WASP bootloader)

The FW is tested, working fine. If not, plz contact me at Matrix and Gitter channels.

[61 more replies]