InfiniTimeOrg/InfiniTime #65: Actions Not Working for Me At All - Newbie to Actions

jotasprout, 2020-09-24:
I have no idea what to even say or where to start so I'll just say I've tried multiple times and these are the screenshots I have. Hopefully, they'll be more helpful than I am. I have no idea which ones will or will not be helpful.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

lupyuen, 2020-09-24:
Hi: Sorry for the delay... The error in the log is this...

https://github.com/jotasprout/Pinetime/runs/1138700180?check_suite_focus=true

```
/home/runner/work/Pinetime/Pinetime/src/drivers/TwiMaster.cpp:1:10: fatal error: sdk/integration/nrfx/nrfx_log.h: No such file or directory
 #include <sdk/integration/nrfx/nrfx_log.h>
```

Could you please edit...

https://github.com/jotasprout/Pinetime/blob/master/src/drivers/TwiMaster.cpp

And change the first 2 lines from

```c
#include <sdk/integration/nrfx/nrfx_log.h>
#include <sdk/modules/nrfx/hal/nrf_gpio.h>
```

To...

```c
#include <nrfx_log.h>
#include <nrf_gpio.h>
```

And it should work, hopefully.

I wrote about this bug here...

https://lupyuen.github.io/pinetime-rust-mynewt/articles/cloud#our-first-pinetime-firmware-build

Under the section "4. Our First PineTime Firmware Build", paragraph "If we see this error...". 

I'm terribly sorry my instructions were not clear, please let me know whether this fixes the build. Thanks.

jotasprout, 2020-09-27:
Your instructions were clear. I just didn't see the error when I was looking because it was deep into that line.

closed 2020-10-04 by JF002