InfiniTimeOrg/InfiniTime #70: Bluetooth connection drops after a while

Avamander, 2020-10-02:
For some reason I see the Bluetooth connection dropping after a while. I haven't determined the exact reason, but it should be taken a look at at some point.

Avamander, 2020-10-11:
Yep, after a while it just seems to disconnect. For some reason the PineTime itself thinks the connection is alive, might need a bit better status polling/handling. Needs to be investigated more.

JF002, 2020-10-28:
Here is a first analysis:
 - The class `NimbleController` should be refactored and improved (better state machine for service and characteristic discovery, handle the return codes of all functions from NimBLE, correctly de-init() all objects when disconnecting,...).
 - Some connection issue might be caused by handles that are not cleared on disconnection. We it reconnects to another device/app, it still use old handles values that are not valid anymore.
 - If NRFConnect and Gadgetbridge are installed on the smartphone, it seems that the CTS from NRFConnect confuses InfiniTime when connecting to gadgetbridge : InfiniTime/NimBLE find a reference to the CTS service during discovery, and fails when reading the time characteristic -> the error triggers a disconnection.
- *sometimes*, NimBLE looks totally dead, advertising does not start anymore, only a system reset solves the issue

I still have to analyze this a bit deeper. In the mean time, I've observed that disabling the CTS service from NRFConnect improved the connection and reconnection stability for Gadgetbridge.

JF002, 2020-10-28:
Here is a first fix : the services are reset on disconnect, and advertising is not started if a connection is already established : https://github.com/JF002/Pinetime/tree/improve-ble-connection
With this fix, and if the CTS service of NRFConnect , connection and reconnection worked every time I tried.

JF002, 2020-10-29:
I've just pushed a new commit. I refactored the service discovery procedure. It' now much cleaner and it does not try to get the time from CTS when it has not been discovered correctly.

There is still one issue, but I don't think it comes from InfiniTime : if the CTS service is enabled in NRFConnect when connecting using Gadgetbridge, CTS is discovered but does not respond to a read command, a timeout occurs which trigger the disconnection of the BLE connection...

@Avamander Can you try this branch and see if it improves the situation on your side?

Avamander, 2020-10-29:
@JF002

> Can you try this branch and see if it improves the situation on your side?

Can it wait until the middle of next week?

Avamander, 2020-10-29:
Also, just wondering, did you test Out Of Range behavior?

JF002, 2020-10-29:
> Can it wait until the middle of next week?

We are not in a hurry. I would like to merge this in InfiniTime 0.9 is it works correctly.

> Also, just wondering, did you test Out Of Range behavior?

Yes, I did many tests, going out of range with my phone, disconnect the bluetooth adapter from my computer while it's connected, connect from NRFConnect, Gadgetbridge and my computer,...
As long as the CTS is disabled on NRFConnect, it works fine.

JF002, 2020-10-30:
I've just created the PR (https://github.com/JF002/Pinetime/pull/118). I'll merge it in develop and create a release candidate release for people to test it

ashkitten, 2021-01-01:
i'm having this problem on the latest release 0.9.0 with a sealed pinetime. after a while (some hours, it seems), it disconnects but still thinks it's connected (so can't reestablish connection) until a reboot.

Avamander, 2021-02-14:
https://github.com/JF002/Pinetime/releases/tag/0.13.0 was just released, but it will take some time to monitor this

JF002, 2021-03-25:
I think we can say this bug was fixed in 0.12/0.13, what do you think @Avamander  ?

Avamander, 2021-03-25:
@JF002 Probably.

closed 2021-03-25 by Avamander