InfiniTimeOrg/InfiniTime #72: The time should be updated before waking up

Avamander, 2020-10-02:
The time should be updated before waking up. Right now it quickly flickers, this wastes a tiny amount of time and energy drawing the wrong number, this could be improved primarily for just polish.

petterhs, 2021-04-20:
I don't think this is the case anymore. Could be closed

christianfl, 2021-05-19:
I can confirm that for the digital clock @petterhs (which was afaik the only watch face when this issue was created). However, the analog watch face introduced in InfiniTime 0.15.0 does show the old time for a few ms after waking up the clock. The amount of time is so little that I couldn't record it with my smartphone but it is there and noticeable. Happens with InfiniTime 1.0.0.

tmilburn, 2021-08-19:
I think this is due to the LCD being woken up before the current app screen is refreshed. The LCD's frame memory appears not to be erased during a soft or hard reset which causes it to display the last screen when switched on. The app then refreshes the screen but in the case of the analog watch face this takes 10s of milliseconds causing the last screen to be briefly visible.

A possible fix would be to run a single refresh of the current app screen before turning on the display such that the LCD's frame memory is updated.

Riksu9000, 2021-08-19:
> I think this is due to the LCD being woken up before the current app screen is refreshed. The LCD's frame memory appears not to be erased during a soft or hard reset which causes it to display the [...]

It is most likely caused by something like this as #586 seems to fix this issue.

tmilburn, 2021-08-19:
I'm surprised that #586 fixes the issue because that PR only removes unnecessary commands to the LCD. The display still comes on before the app is refreshed.

Riksu9000, 2021-09-04:
The display takes some time to wake up after the Sleep Out command and it displays two blank frames. I think we're intended to update the screen during this blank period, but the commands added so much delay that it couldn't update the display in time, so it displayed what was left previously. Therefore I think this issue can be closed with #586.

watchdominiondotcom, 2021-09-28:
the time does not sync it says 1978. does it just not work or did i not figure it out?

JF002, 2021-12-27:
Latest versions of InfiniTime (1.7) improved many aspect of the UI and UX. I think we can close this issue. Feel free to reopen if needed.

closed 2021-12-27 by JF002