InfiniTimeOrg/InfiniTime #76: Potential Bluetooth failure after a failed DFU

Avamander, 2020-10-03:
I think I just stumbled upon a state where a DFU update fails, it returns to the main screen, but I couldn't establish a Bluetooth connection any more. Rebooting the watch fixed it. 

This needs investigation because it's very bad with sealed PineTimes

Avamander, 2020-10-03:
Just managed to make it happen again, so I'm pretty sure it's an issue.

davorvr, 2021-01-08:
I can confirm this with 0.10.0, failed DFU update makes the watch unreachable unless I reboot it.

drpong2, 2021-01-08:
I've also encountered this issue on 0.9.0, but don't recall having the issue on 0.10.0 or with the RC3 bootloader.

I can do further testing and confirm.

Avamander, 2021-02-14:
https://github.com/JF002/Pinetime/releases/tag/0.13.0 was released, we'll have to see if that fixed it.

petterhs, 2021-04-20:
> https://github.com/JF002/Pinetime/releases/tag/0.13.0 was released, we'll have to see if that fixed it.

I have not had this issue since 0.13.0. Could be because OTA failures are rare now, but I think it is fixed

MysteriousLog6, 2021-06-09:
Prithee closeth this issue

JF002, 2021-06-10:
With pleasure :)

closed 2021-06-10 by JF002