InfiniTimeOrg/InfiniTime #87: Littlefs (or similar) datetime backup

Avamander, 2020-10-06:
It would be nice if the watch kept track of at least the current date on it's own. 

Writing a new date to a file once a day should be okay in terms of wear, but it'd make the watch much more useful if it needs to reboot in a situation where there's no phone nearby. Plus it'd make date look nicer during testing which constantly resets it.

JF002, 2020-10-07:
Yes, good idea! We can also write the current date/time to flash just before issuing a reset (after a successful OTA, for example).

ZephyrLabs, 2020-10-08:
I have a proposal, perhaps we can create a partition in the FS for the SPI flash such that is some sort of EEPROM emulation for non-volatile data, eg. the request for time backup by @Avamander or user settings and other future things... 4Kb of storage should be ample for such a thing..

JF002, 2020-10-08:
There's no FS in the SPI flash for now. The first 115200KB are used by the boot graphics, the rest is empty. As @Avamander wrote, I think the best would be to use a embedded FS like littleFS to manage the reste of the SPI flash space. 
I think littlefs provides a higher level and easier to use API than a simple EEPROM.
We would be able to use this FS to store configuration parameters, data time, graphic assets, heart rate values,...

Avamander, 2020-10-08:
> There's no FS in the SPI flash for now. 

@JF002 Working on that right now :)

AirHamster, 2020-11-04:
What about RAM retention?
We can store time in a certain part of RAM that will be excluded from startup initialization.
And after reboot time can be restored from this part of memory.
https://devzone.nordicsemi.com/f/nordic-q-a/272/ram-banks-ram-retention

MysteriousLog6, 2021-05-04:
> What about RAM retention?
> We can store time in a certain part of RAM that will be excluded from startup initialization.
> And after reboot time can be restored from this part of memory.
> https://devzone.nordicsemi.com/f/nordic-q-a/272/ram-banks-ram-retention

Ram is volatile memory right?

Avamander, 2021-05-04:
> Ram is volatile memory right?

Yes. There are sleep modes where RAM contents are kept (with extra power used) and methods to keep some amount of RAM untouched.

MysteriousLog6, 2021-05-05:
Well what if the device runs out of battery?

pfeerick, 2021-05-12:
If it runs out, it looses it... simple as that. That's when you would revert to the datetime stored in flash memory. But for reboots, either manual or update related, storing it in RAM would be perfect as it would save a write to the flash memory and preserve the time across the reboot (bar the bootloader/DFU update delay).

myxor, 2021-06-15:
This could not only be used for date and time but also for other data like steps counter and last hrm result.

MysteriousLog6, 2021-06-16:
> This could not only be used for date and time but also for other data like steps counter and last hrm result.

I think there is a discussion for this , we could of course do this for a whole lot of other things.

kieranc, 2021-10-01:
This has been implemented in #595 and can be closed.

closed 2021-10-02 by Avamander