InfiniTimeOrg/InfiniTime #88: Github Simulate Action fails but Build Action succeeds

hrmckay, 2020-10-06:
I followed lupyuen's instructions for building the Github Actions to Simulate and Build InfiniTime.  I made a change to clock.h to add a new variable "lv_obj_t* dayofweekday;" and I made changes in clock.cpp to use that variable.  When the actions ran the Simulate action failed but the Build action worked.  The Simulate action failed with "error: use of undeclared identifier 'dayofweekday'".  It looks like the Simulate action is not using the clock.h in my repository.

Hugh

JF002, 2020-10-07:
Could you give us the link to your repo so that we can have a look at the code?
Note that I'm not using Github Actions, and I might not be able to help you, but I hope someone else will be able to help you debug this issue.

hrmckay, 2020-10-07:
Since lupyuen set up these actions I think he is probably the best person to answer this, but I couldn't see any way to post an issue to his documentation.  I am using https://github.com/hrmckay/Pinetime

Hugh

lupyuen, 2020-10-08:
Hi @hrmckay: The InfiniTime Simulator runs in a "Sandbox" and only changes to Clock.cpp are accepted. No other files may be changed. 

A workaround is to declare `dayofweekday` as a static variable in Clock.cpp.  I'll update this in the doc, thanks for raising this.

BTW you can submit a PR for my doc here...

https://github.com/lupyuen/pinetime-rust-mynewt/blob/master/rust/app/src/simulator.md

lupyuen, 2020-10-08:
@ZephyrLabs We spoke about marking the sections in Clock.cpp, now might be a good time to revisit this :-)

```c
//  Define Your Widgets Here
...
//  Create Your Widgets Here
...
//  Refresh Your Widgets Here
...
```

ZephyrLabs, 2020-10-08:
@lupyuen I shall submit a PR for marking those regions.
but I believe the section could be more broadened with,
 1. normal refresh, using the `Clock::refresh()` function Independent of time change... 
(useful for things such as animations)

 2.Refresh based upon time change 
(useful for things like the Analog clock(future), AM/PM time Indicators etc.)

lupyuen, 2020-10-08:
@ZephyrLabs Thanks! :-) Have a chat with JF... See how best we can guide our newbies

ZephyrLabs, 2020-10-08:
No problem, please tag me, I will respond as soon as possible :-)

lupyuen, 2020-10-08:
FYI I'm working on a new Simulator for Rust on Mynewt... Hopefully the Rust Compiler will do better validation and prevent such Sandbox problems.

ZephyrLabs, 2020-10-08:
@JF002 which branch can I leave a PR for the marked sections of Clock.cpp ?
I have a opinion of rearranging the entire Clock.cpp so that its both functional and acts as a barebones design template for future designs...

hrmckay, 2020-10-08:
Thanks for the information.  I should have read the documentation more carefully.  I was going to close this issue but you seem to be using it as a message board so I will leave it open.

Hugh

JF002, 2020-10-08:
@ZephyrLabs Please submit your PRs to the **develop** branch (https://github.com/JF002/Pinetime/blob/master/doc/contribute.md). Thanks!

ZephyrLabs, 2020-10-09:
@JF002 I have added commented regions for addition of new widgets, Pull request to Develop Branch created, please continue the topic from there, Thank you! #90

lupyuen, 2020-10-09:
Thanks that should be helpful :-)

ZephyrLabs, 2020-10-09:
welcome :-)

closed 2020-10-11 by JF002