InfiniTimeOrg/InfiniTime #89: Can I haz Hacktoberfest, please?

jotasprout, 2020-10-06:
Based on a tutorial I followed and help from @lupyuen on issue #65 (mine, now closed), I'd like to contribute to this repo and his (and perhaps others) as I am absolutely loving learning and hacking with PineTime. I'm also passionate about Hacktoberfest so, if you're willing, could you opt in to it? If not, I still want to contribute and definitely will, but, if you're willing, here's some info:
[image]

lupyuen, 2020-10-06:
@JF002 What do you think?

Avamander, 2020-10-06:
@lupyuen the topic is already applied.

JF002, 2020-10-07:
Yes, the label is already applied on the repo. Do I need to do something else to enable hacktoberfest?

Avamander, 2020-10-07:
You don't have to do anything else.

closed 2020-10-11 by JF002