InfiniTimeOrg/InfiniTime #91: Screen update bugged after certain hardware scrolling stuff has happened

TT-392, 2020-10-09:
his was an issue I noticed when I was testing Electr0Lyte's pong code. If you start the watch, go to the validation screen, and scroll down. Then on (I am guessing 80 horizontal lines) the screen isn't updating correctly and causing weird stripes to form on the screen when the ball crosses.

I am guessing this is related to the screen being left in a scrolled state after having left this version validation screen, and this is somehow not correctly handled when writing stuff to the screen.

I noticed this with pong, but I am pretty sure that other weird graphical bugs on other apps where stuff is moving are related
[image]

Avamander, 2020-10-11:
Can confirm the same happens when mid-scroll a DFU update starts, a full display scroll might need a lock or the display some kind of transaction buffer.

JF002, 2021-03-25:
Fixed in 0.15

closed 2021-03-25 by JF002