InfiniTimeOrg/InfiniTime #94: Upgrade to LVGL-7

ZephyrLabs, 2020-10-10:
I propose this thread to create a new branch of Pinetime for updating the current LVGL6 to LVGL7

JF002, 2020-10-11:
As upgrading from one major version to another is not a trivial change, I think we must ensure that the project really need to upgrade to LVGL7. Are there any features or improvements that could be useful for InfiniTime?

It could also be useful to check the memory (flash/ram) usage and the rendering time of the firmware using lvgl6 so that we can check if the upgrade actually improve (or deteriorate) the performances of the firmware.

Avamander, 2020-10-11:
But about the changes (see: https://github.com/lvgl/lvgl/releases?after=v7.0.1):

* There's `lv_img_set_zoom`, `lv_img_set_angle` and `lv_img_set_pivot` which sound really nice.

* Better various locale support

* Bugfixes

* I think also better type-ing

* Charts

* More bugfixes

Plus: 

* It's a good opportunity to turn the library into a git submodule for better dependency handling.

* It's easier now, rather than later.

We might even want to consider skipping 7 and just go with beta 8 though in one burst...

ZephyrLabs, 2020-10-11:
@JF002 perhaps we can take a copy of the develop branch and use it specially for testing LVGLv7 or beta 8

JF002, 2020-10-11:
@ZephyrLabs Right, we can create 2 feature branches to test both versions.
@Avamander Thx for the list  of new features, I can see there are many advantages in this new version! I'm not used to git submodules, do you know if to use them properly?

Avamander, 2020-10-11:
@JF002 
> I'm not used to git submodules, do you know if to use them properly?

Git submodules are basically Git repositories in Git repositories. If you already clone a repository into your repository, it should probably just be a Git submodule. In a sense it does what dependency version locks do with things like `pip` or `npm`, but just with Git repositories. It's very nice and much more neater than manually keeping copies of some other repo in your own.

Scott Shawcroft who maintains and writes a lot of code for Adafruit uses them extensively just talked about them in his livestream: https://www.youtube.com/watch?v=vAPOZrG8pBo&t=1332s

JF002, 2020-10-11:
There is something wee must keep in mind when upgrading lvgl: We use a slightly modified version of lvgl to add support for the HW scrolling feature of the display controller. 
Thanks to [Koen](https://github.com/bosmoment) for his [patch](https://github.com/JF002/Pinetime/tree/master/src/libs/lvgl/patches) !
I don't think this change has been streamlined so we might need to port it to lvgl 7 and 8 (and it would be better if we tried to merge this modification to lvgl directly...).

bergzand, 2020-10-11:
>  Thanks to [Koen](https://github.com/bosmoment) for his [patch](https://github.com/JF002/Pinetime/tree/master/src/libs/lvgl/patches) !

You're welcome!

I took a quick look, doesn't seem that the refresh algorithm (a for loop) changed during the v6 → v7 upgrade. I think the patch should still apply, maybe with minor modifications.

A submodule should simplify your work flow a lot here. You can keep a personal fork of LVGL, with the patches you need applied to a branch. The submodule can then use that fork/branch. Updating LVGL is as simple as creating a new commit/branch and updating the submodule reference.

bergzand, 2020-10-11:
[This is the patch](https://github.com/bosmoment/RIOT/blob/bfe228cb8c5dc5418c0707c26804a49a90629d21/pkg/lvgl/patches/0001-lv_refr-add-support-for-selecting-render-direction.patch) in case anybody here wants to try it out on a new LVGL version

Avamander, 2020-10-11:
> A submodule should simplify your work flow a lot here. You can keep a personal fork of LVGL, with the patches you need applied to a branch. The submodule can then use that fork/branch. Updating LVGL is as simple as creating a new commit/branch and updating the submodule reference.

Yep, this is a possibility as well.

JF002, 2021-03-25:
LVGL7 was upgraded in 0.14 :)

closed 2021-03-25 by JF002