InfiniTimeOrg/InfiniTime #97: Music: long titles move back and forth too fast

arteeh, 2020-10-11:
0.8.2

When playing a song with a title that's longer that the Pinetime screen can draw (about 18 or 19 characters?), the title moves back and forth extremely fast, making it nearly unreadable. 

Expected: The title moves back and forth slowly, so it's easy to read.

Ideal: The title moves forward, then when it's at the end, it immediately goes back to the front. (People don't read backwards so text moving backwards is unintuitive)

arteeh, 2020-10-11:
This may be fixed with #96

Avamander, 2020-10-11:
It is not fixed in the PR, it's something in LVGL. That's a topic for an another PR.

jedmijares, 2020-12-30:
I took a shot at adjusting this in #152, here's a video showing what it looks like: https://www.youtube.com/watch?v=PvBWN1-NBNU

PR #259 (2021-04-09): Fix music app
 - Enable LVGL animation (and disable groups, which were not used), and set the speed.
 - Fix disc animation and progress display by initializing lastIncrement at 0 (a random value will be used otherwise, in release build)

Fix #97  and #130
closed 2021-04-09 by JF002