InfiniTimeOrg/InfiniTime #102: Suggestion: make a % 24h circle - sphere - watch

mitcoes, 2020-10-15:
Some time ago Swatch had the idea of Swatch time, that divided the day in 1.000 beats and 2 decimals with the GMT+1 hour

https://en.wikipedia.org/wiki/Swatch_Internet_Time

A sphere clock that would do almost the same but with % and 24h and sun and moon rise / down, rains, waves, winds, temperatures and any other concentrically circles you would add would be great to have to check not only time at one sight. 

If you make it also available in a 12h or even a more detailed 6h format if not much used would be a good PR trick that can help to sell more units to main stream people and improve the project.

Avamander, 2020-10-15:
This would be a good beginner project. Are you sure you don't want to try and implement it yourself?

JF002, 2020-10-18:
Thanks for this idea! As Avamander said, that would be a cool project if you want to do it !
Could you maybe provide a drawing of your concept so people can get some inspiration if they want to build this watchface?