InfiniTimeOrg/InfiniTime #104: Add a changelog for versioned releases

oholiab, 2020-10-18:
Hello!

It would be really helpful to be able to see the changes between released versions to be able to know when critical bugs are fixed and new features are added without having to read over all of the commits in between two tags.

This will be especially useful as more users start to come along and want to know whether features are in their shipped version and whether they need to upgrade.

Avamander, 2020-10-18:
All the releases already contain a changelog.

oholiab, 2020-10-18:
Oh sorry, I didn’t realise that was in the tag metadata!

I’ve worked too much on Debian packages and having to hand hack the changelog file!

Sorry for being confused all over the issues :S

closed 2020-10-18 by oholiab