InfiniTimeOrg/InfiniTime #107: Set up CI to test all build targets

Avamander, 2020-10-20:
It would be very useful to build all the possible targets (and flag combinations) when PRs or commits are made to the repository. This helps avoid mistakes.

If someone knows GH actions or some other CI service well enough, it'd be a nice contribution.

JF002, 2020-10-21:
[Github Actions](https://github.com/JF002/Pinetime/actions) are already enabled on the repo, but I have to admit I don't know exactly how it works. They have been mainly contributed by @lupyuen.

My "dream" CI is something that is easy to deploy and run on any computer or server. Most CI tools I've found are either SaaS or unreasonably complex for a single project written in a single language.

There is a [docker image](https://github.com/JF002/Pinetime/blob/develop/doc/buildWithDocker.md) that builds the whole project and generate all artifacts (bootloader, images, dfu,...). It just misses a bit of automation (automatically run when a new commit is pushed) and maybe a web frontend.

maidenone, 2020-10-23:
you can soon run on my test nodes with the test API.

python3 runOnTestNode.py [testFile] [testNode] [fw.hex]

currently it does not return anything, but i will add a "job id" so you can get test results once the job is done. for now you have to manually check the test results...

luben93, 2020-11-05:
is this still an issue, if so i would like to help i am comfortable with CI workflows, and since my pinetime is broken i cant really help on more hardware related issues

JF002, 2020-11-05:
Yes, I think we can still work on this ! For now, we have Github Actions (people contributed to them, but I don't use them very often), a docker image and the testbench from @maidenone.

What do you have in mind for the CI?

maidenone, 2020-11-05:
i can open up the ports needed for the comandline API tool to work, then every github action build can also trigger a test execution on real hardware.

in the future it will even be possible to link to test results and see pass/fail. for now you have to manually go to http://foss.site to see the results.

JF002, 2020-11-05:
Wow, a distributed C/I for PineTime, awesome!

luben93, 2020-11-13:
so i dont really have a concrete plan i was just thinking that i'd take the tools that you like working against the most and make them build on every commit in a PR for example

luben93, 2020-11-13:
im currently (albeit slowly, whenever i have time after work) looking into making the dockerfile without having to build it for 5+ minutes on my machine, i dont even want to know how slow its on arm, so that the docker container can just be downloaded and executed to build the code, which suites a CI workflow

closed 2021-12-30 by JF002