InfiniTimeOrg/InfiniTime #113: API for plugins?

denizzzka, 2020-10-24:
Hi!

It is possible to add plugins API or ABI? For example, for add some more external binaries during linking.

Perhaps someone is interested in writing add-ons for Pinetime on other languages ​​than C++ and with a different than gcc compiler.

(I am personally want to try D language with ldc2 compiler based on LLVM)

Avamander, 2020-10-24:
It's possible, but nobody has spent the effort do so. Feel free to try.

JF002, 2020-10-27:
External apps and stable API/ABI are something we'll want to work on one day or another! To be honest, at this time, I have no idea how we'll build binaries that we'll be able to load dynamically into the running firmware. 
As @Avamander said, if you want to do some researches and do some experiment, feel free to share your results :)

denizzzka, 2020-10-28:
> I have no idea how we'll build binaries that we'll be able to load dynamically into the running firmware.

For a beginning static linking is enough.

JF002, 2022-08-02:
I've just created a more generic feature request about dynamic app and custom watchface to group multiple requests around that topic. Feel free to continue the discussion in #1262 ;-)

closed 2022-08-02 by JF002