InfiniTimeOrg/InfiniTime #115: Feature request: Haptic alarm app

Izaic, 2020-10-27:
It would be awesome to have a haptic alarm app that lets you set an alarm time and then starts vibrating the watch till you tap the screen to confirm that your aware of the alarm.

JF002, 2020-11-19:
Hi @Izaic ! Good idea, indeed! 
Could you describe your idea with a bit more detail? Maybe a UI drawing? Also, how would you set the alarm time? Using the watch directly, or using a companion app ?

vxsbh, 2020-11-19:
I think having the ability to set the time on the watch itself would be more practical 🤔

Izaic, 2020-11-19:
> Hi @Izaic ! Good idea, indeed!
> Could you describe your idea with a bit more detail? Maybe a UI drawing? Also, how would you set the alarm time? Using the watch directly, or using a companion app ?

[image]

I stole this from the waspos issue i created on adding a haptic app to waspos as well... As you can see, it's just a plain and simple idea... Tap what time you want it to go off, press a button to confirm the time, and when the time is reached it starts vibrating till to adknowledge it (likely by having a little popup that says "Press ok to stop haptic alarm")

Izaic, 2020-12-15:
Been using the Wasp-OS haptic alarm app for a few days now and i have to say its an amazing feature! The vibrator in the pinetime is stronger than i though (funnily enough it's stronger than the pinephones)

AnirudhAdiraju, 2021-01-01:
Hello Izaic!
I have actually finished an alarm app, but there are some issues I need to fix with the logic. I would love for you to check out a demo of it when I get my pinetime up and running again

ncartron, 2021-02-16:
Agreed, an alarm would be nice - I usually use my watch to wake me up in the morning, and that's definitely one thing which I'm missing right now on my PineTime :)

AnirudhAdiraju, 2021-02-16:
Would you prefer an alarm set for a certain date and time or just a time and it rings at the nearest instance of that time?

Izaic, 2021-02-16:
> Would you prefer an alarm set for a certain date and time or just a time and it rings at the nearest instance of that time?

Waspos implements just a time. I guess it doesn't matter though. Would be neat to be able to set different dates with different times.

ncartron, 2021-02-17:
yes - also, if I were to ask for a "perfect alarm app", recurring alarm (e.g. every day at 7h) would be super nice :)

jave, 2021-08-03:
I have a pinetime and a fitbit sense. I switch every night to the Sense so it can wake me up, and I switch back to the pinetime in the morning.

The thing I like with the sense alarm: 
- It wakes me every work day, at the same time, but not at weekends (configurable of course)
- Theres a big button on the display I can snooze the alarm, and a smaller button I can dismiss it with, this is useful.

Its convenient to manage the alarm on the watch itself.

It would be nice not to need switching back and forth between the pinetime and the sense, and this is the only feature making me do so. All the other sense features missing in  pinetime are less important, at least IMHO.

Riksu9000, 2021-10-31:
An alarm app has been included since 1.5.0

closed 2021-10-31 by Riksu9000