InfiniTimeOrg/InfiniTime #116: Date goes off the right side of the screen

Izaic, 2020-10-28:
The date WEDNESDAY 28 OCT 2020 goes of the screen on the right side with the 2020 clipped so that only 202 shows.

thelovebug, 2020-10-31:
Might I suggest that if the month is being abbreviated, then the day should be also.

Having said that, today: "SATURDAY 31 OCT 2020" fits perfectly and looks amazing.

JF002, 2020-11-01:
Mhm when I implemented this, the whole date was displayed... It was in 2019, I guess!

x2k13, 2021-03-16:
Suggestion:
I don't know if this is possible, since I have no idea how to code that, but how about setting a slightly smaller font size and use  center alignment instead of left alignment. This would make sure the date is always in the middle and not cut off to either of the sides. What do you think?

JF002, 2021-03-25:
Fixed in 0.15

closed 2021-03-25 by JF002