InfiniTimeOrg/InfiniTime #117: Notification popups do not show enough text

Izaic, 2020-10-28:
The notification popups are just too tiny to see enough text. In my opinion they should take up the whole screen and close after a few seconds, or require a press to confirm that you've seen the message.

Avamander, 2020-10-28:
Will be improved by https://github.com/JF002/Pinetime/tree/notification-ui maybe available in the next release

JF002, 2020-11-01:
If you have a dev kit with an SWD debugger, you can try this release candidate version and tell us what you think : https://github.com/JF002/Pinetime/releases/tag/0.9.0-develop

Warning : Do not try it on a sealed device!

JF002, 2020-11-19:
The new notification UI is now available in InfiniTime 0.9.0 and is supported by gadgetbridge 0.49! I'm closing this issue.

closed 2020-11-19 by JF002