InfiniTimeOrg/InfiniTime #119: Battery level always shows 100% or 0%

thelovebug, 2020-10-31:
Firmware 0.8.3 running on a sealed PineTime

The battery level on the information screen (bottom left icon on the first menu) always shows 100% or 0%.  I can't tell you at what point the level changes, but it was showing 100% for at least a day before changing to 0%, and it's been at least 3 days since then.

The battery icon on the main time view seems to reflect the battery level correctly, however.

[image]

JF002, 2020-11-01:
Thanks for your feedback! I'll have a look when I have a bit of time ;)

IoTPanic, 2020-11-02:
+1 Same issue, battery state in the battery icon on home seems fine, however after updating from 0.7.1 to 0.8.3 broke the menu battery percentage as pictured in thelovebugs post. Hope this tad extra info helps.

commit 3703611a (2020-11-19): Fixes #119! Improved based off suggested changes
closed 2020-11-19 by JF002