InfiniTimeOrg/InfiniTime #121: Enhancement - Email

SecureCodeJIT, 2020-11-02:
request ability to connect to various email systems such as hotmail, gmail, etc.

JF002, 2020-11-04:
This device (hardware) only support BLE connectivity. There's no Wifi radio on the pinetime, it won't be able to connect to the internet to fetch your emails.
However, there is notification support. Apps like Gadgetbridge and Amazfish can send notification to the pinetime when your phone receives an email.

JF002, 2020-11-19:
@SecureCodeJIT Thank you for your enhancement ideas. However, we'll need more info on how you think these features should be implemented (more details on the UI, functionalities, integration with companion apps, examples on other existing products,...).

I'm closing this as there were no update for the last 15 days, but feel free to reopen it if you want to discuss your ideas with me and the other contributors .

closed 2020-11-19 by JF002