InfiniTimeOrg/InfiniTime #122: Enhancement - Interface with Health Apps

SecureCodeJIT, 2020-11-02:
Provide ability to link / pass data to other health apps.

JF002, 2020-11-04:
Do you have a specific app in mind? And does this app provide any documentation about its API?
Please keep in mind that the heart rate sensor and motion sensor are not integrated in the current state of the project .

JF002, 2020-11-19:
@SecureCodeJIT Thank you for your enhancement ideas. However, we'll need more info on how you think these features should be implemented (more details on the UI, functionalities, integration with companion apps, examples on other existing products,...).

I'm closing this as there were no update for the last 15 days, but feel free to reopen it if you want to discuss your ideas with me and the other contributors .

closed 2020-11-19 by JF002