InfiniTimeOrg/InfiniTime #127: Broken song current length when re-entering music app.

AirHamster, 2020-11-06:
Pinetime: 0.9.0-dev
Gadgetbridge: 0.48
Android: 10

If you start playing from music app, then click side button and enter music display again, the current playing time will be wrong. And it is stays still.

It looks like this time the same as uptime.

[image]

[image]

JF002, 2020-11-07:
From your description, the issue might come from [here]https://github.com/JF002/Pinetime/blob/develop/src/displayapp/screens/Music.cpp#L189 : When you quit and re-run the Music app, the variable `lastIncrement` is probably initialized to 0, which would explain why the display time corresponds to the uptime of the device.

Avamander, 2020-11-08:
I'll try to submit a patch for the next release.

pfeerick, 2020-11-17:
A possible secondary behaviour is one where the elapsed time stays at 00:00 until you pause the song, either from the watchapp or from the phone. As soon as you do this it 'catches up' and is correct from then onwards. Pinetime: 0.9.0-dev and Gadgetbridge: 0.48 also.

Avamander, 2020-11-17:
@pfeerick No, that specifically is a limitation of your music app, it simply doesn't send the needed information until some event is triggered.

pfeerick, 2020-11-17:
Interesting that you would just jump to the conclusion that it it must be a limitation of the music app, without asking *which* app it is, and do I have any other devices to compare it to. I get the same behaviour with VLC and Spotify, but not if I use my Pebble watch. It updates perfectly, whereas Gadgetbridge + the music app do not. Nor does changed track progression through seeking get detected (which it does on the Pebble). Perhaps it's a limitation in _Gadgetbridge_, since I'm using the Pebble app with that?

Avamander, 2020-11-17:
@pfeerick because I know it's an universal limitation, the app doesn't matter. There's no way for a music app to know it needs to re-broadcast the track info intent.

Seeking updates the progress of the song assuming your music app sends the necessary intent.

pfeerick, 2020-11-17:
Perhaps you didn't read my last comment fully. I am using two watches - infintime and pebble (classic) and one works fine with two different music apps, and the other doesn't. 

* Infinitime watch Music app - does not update when seeking, and often when starting playback from it will not start counting elapsed time, unless you pause/play again, from either the phone or the watch. With Gadgetbridge, and Spofity or VLC

* Pepple watch Music app - DOES update when seeking, and when starting from it - no pause/play required. With Pebble app, Spotify or VLC. 

Thus it isn't the music app, because the Pebble would have the same problem, which it does not. If it were the vendor bundled music app, I would agree with you, because it does have issues generally, for both, but this is not what is being spoken about.

Avamander, 2020-11-17:
@pfeerick perhaps you haven't seen this video https://youtu.be/YvER1JsuPOg?t=26 - progress is updated on seek assuming your player sends the information. It uses the very much the same code path for receiving events as Pebbles do. 

Thus it is something about your setup that's weird.

pfeerick, 2020-11-17:
Thank you, I had seen that video, but it reminded me to try in PowerAmp as well, to see if that behaves differently, which it does. 

So, the Infinitime+Gadgetbridge+Poweramp combo works flawlessly from that I can tell, but Infintime+Gadgetbridge+[VLC or Spotify] does not. But Pebble+Pebble app+[PowerAmp or VLC or Spotify] also works perfectly, suggesting the issue is in the companion/bridging app, not the music app on the phone. 

The Pebble watch music app would not be able to update to show the seek if the information was NOT being made available to the companion app. Perhaps apps like VLC and Spotify are not presenting it to GadgetBridge in a form it wants (or there is yet another bug in it).

JF002, 2020-11-19:
If I understand correctly, this is more an issue on the companion app (or music app or smartphone or ...) side than on InfiniTime side?

pfeerick, 2020-11-19:
Yes, these last few comments have come down to being issues with the companion apps, not InfiniTime. The first three are still relevant as the issue hasn't been patched yet.

JF002, 2020-12-01:
So we still have to try to fix this : https://github.com/JF002/Pinetime/issues/127#issuecomment-723422720 :)

PR #1057 (2022-04-25): Update track progress in MusicService
Update the progress in the music service, instead of the app.
I'm unsure if this is the best way to fix it, and this probably warrants more thorough testing, but in my limited testing everything seems to work correctly.
Fixes #127
commit 34858d0a (2022-04-25): Update track progress in MusicService. Fix #127
closed 2022-04-25 by Riksu9000