InfiniTimeOrg/InfiniTime #129: OTA Broken after install of 0.9.0

kgj, 2020-11-08:
Not sure what the cause is yet but OTA firmware updates are not working after pushing 0.9.0

Error:

================================
    ==                            ==
    ==         DFU Server         ==
    ==                            ==
    ================================
    
Sending file image-0.9.0.bin to D7:2C:58:C7:1C:8C
Binary imge size: 268208
Binary CRC32: 2078034561
Connecting to D7:2C:58:C7:1C:8C
Need to switch to DFU mode
Exception at line 147: UUID not found: 00001531-1212-efde-1523-785feabcd123
DFU Server done

JF002, 2020-11-08:
Hello!
The good news is that OTA is **not** broken in 0.9.0, it still works with NRFConnect and Gadgetbridge.

However, I confirm that it does not work using the script dfu.py. I'm not sure why it can't find the characteristic *00001531-1212-efde-1523-785feabcd123*... NRFConnect is able to discover it without any issue.

JF002, 2020-11-08:
I've just push a new commit to *dfu.py* : https://github.com/JF002/Pinetime/commit/f0e1f98823e41bfc2d9743fa8de70c882f26f93b (last commit on develop)

I've increased the timeout of characteristics discovery, and now, the script is able to initiate the OTA. @kgj could you please confirm it works for you?

kgj, 2020-11-08:
Hi There,

Thanks for the quick fix! 

I can confirm that OTA is working again.

closed 2020-11-08 by kgj