InfiniTimeOrg/InfiniTime #130: Music app, track text scrolls too fast with longer track titles.

dns53, 2020-11-09:
When using the music app on the watch the track artist and track name are displayed.
I have a track name with a longer than will display on the screen so this text is automatically scrolling forwards and backwards.

This seems to be moving the text forwards and backwards one character each frame with this appearing to update more than 20 characters each second making it hard to read.

Could we change the update speed of the text so the text updates at a more easy to read rate such as moving the text forward or backwards a character every second, every half second etc.

Avamander, 2020-11-09:
Duplicate of https://github.com/JF002/Pinetime/issues/97

pfeerick, 2021-05-12:
This can be closed as was fixed in #259.

MysteriousLog6, 2021-06-11:
prithee closeth this issue f'r t doest the not existeth anym're @JF002

closed 2021-10-02 by Avamander