InfiniTimeOrg/InfiniTime #132: Notifications are cut off in notification app

BitWire, 2020-11-10:
InfiniTime: 0.9.0
Gadgetbridge: 0.48
Android: 10

In the new Notification UI/App, every notification is cut off after 18 Characters instead of the 100 Characters as stated in the Changelog.

Example of the bug as shown in the picture. Original notification was 20 Characters (12345678901234567890)
[image]

JF002, 2020-11-11:
I'm sorry, I forgot to mention one little detail : the support for notifications > 18 characters needed a modification in Gadgetbridge. The pull-request has been merged, but not released yet. 
You'll need to build Gadgetbridge from source or wait for the next (0.49?) version of Gadgetbridge to be released.
I'll update the release note with this information.

BitWire, 2020-11-11:
Alright, thanks for making that clear :+1:

closed 2020-11-11 by BitWire