InfiniTimeOrg/InfiniTime #144: Stop watch

tobrie, 2020-11-26:
Like #115 there is also no stop watch yet. I have [attempted to make one](https://github.com/tobrie/Pinetime/tree/feat-stopwatch), but it's very bare bones and is untested since I don't have access to a device.

Izaic, 2020-12-15:
Definitely something that could be handy!

drpong2, 2020-12-28:
This is something I can build on 0.8.3, or do I need to have 0.9.0?

pfeerick, 2021-01-14:
Give it a try and see. Nothing jumps out at me as being tied to a specific version of Infinitime though, but there are several files to be modified in order for the app to be incorporated.

FlameKat53, 2021-03-11:
I will test this once I'm home today

Panky-codes, 2021-03-21:
I think this can be closed now :)

JF002, 2021-03-21:
A new Stopwatch app was added in 0.15.0.

closed 2021-03-21 by JF002