InfiniTimeOrg/InfiniTime #146: Battery level is erratic

hrmckay, 2020-11-28:
In version 0.9.0 and earlier versions I noticed that the battery indicator at the top of the screen would sometimes change erratically.  It would show 25% full, then a few seconds later 50%, then after a few seconds back to 25%, etc.  I changed the clock face to show the actual battery percentage and it changes almost every second.  As I look at it now it shows, 32, 30, 34, 27, 30, 32 changing once a second or so.  This indicates the .IsUpdated() code in Clock.cpp is being executed every second which seems a lot.

Hugh

JF002, 2020-11-29:
The battery level is updated (via an ADC acquisition) periodically, maybe every 1s. No filtering is applied on the value acquired on the ADC, so it's subject to a lot of noise (depending on the actual battery level and cpu load, for example).

Knowing that, I would certainly expect a noisy battery level value. There are several ways to improve that:
 - Probe the ADC when we know that the CPU mostly idle, so that we always measure the voltage on low load
 - filter/average the value to reduce the noise
 - calibrate the %-value : store the min/max voltage value in NVM and use these min/max to provide a better value in %.

Interested to work on that ?

hrmckay, 2020-12-01:
Thanks for the info.  I will look into smoothing the data and let you know what I find out.

Hugh

Panky-codes, 2021-01-20:
I think we can close this issue for now.

hrmckay, 2021-01-21:
I was testing similar code, written in pure C since I have very limited C++ experience.  I got improvements but not as good as I would have liked.  Your code looks excellent and fits in better with the rest of the code.  I will close this issue.

Hugh

closed 2021-01-21 by hrmckay