InfiniTimeOrg/InfiniTime #148: Pressing back button from an InfiniPaint app takes shifted screen

elektro-NIK, 2020-12-06:
Pressing back button from an InfiniPaint app during touch takes shifted screen.

Steps to reproduce:
- enter in InfiniPaint app
- touch and hold
- during holding screen press button

Actual behavior:
- screen be shifted 

[image]
[image]
[image]

Avamander, 2020-12-06:
What InfiniTime version?

elektro-NIK, 2020-12-06:
0.9.0

JF002, 2020-12-07:
Looks like there's an issue with vertical scrolling and InfiniPaint! 
This app bypasses LVGL to write on the display, and might not take the vertical scrolling feature into account. I find the offset a bit weird... the vertical scrolling should always scroll 240px on a 320px surface; If an error occurs, the offset should be 80px. On the picture, it looks like it's less (maybe 20px?).

elektro-NIK, 2020-12-08:
It looks like 10 pixels. If you continue to take these steps, the screen will go full cycle in 32 times

LookinR, 2020-12-16:
The same thing happened, when I used lvgl.FlushDisplay() in constructor of InfiniPaint. (I'm trying to program Snake-game changing the code of InfiniPaint.) Size is the same as FlushDisplay area height. Maybe some issue with time sequence - FlushDisplay inside of InfiniPaint::OnTouchEvent is done after ending application and redrawing the screen...

petterhs, 2021-01-03:
While reproducing this, I tried to make the offset go full cycle and return to normal by reproducing multiple times. When the offset was around 50% of screen height the application froze and the new bootloader reverted to earlier version of firmware.

tmilburn, 2021-08-19:
Appears to be a duplicate issue to #445 but I can't reproduce this in 1.3.0

Avamander, 2021-10-02:
If this still exists ask for a reopen.

closed 2021-10-02 by Avamander