InfiniTimeOrg/InfiniTime #170: watch crashes while using game

C00kie-, 2021-01-17:
Hello, thanks for the project, awesome work!
I was testing a bit the version 0.10.0
I found bug that lead to watch reboot:
- while using the 2*2 puzzle game, after a few turns (maybe 10), graphics froze and watch reboot itself.
- while use drawing, same behavior.

JF002, 2021-01-22:
Hi and thanks :)

Does it happen every time you use these apps? After how long?

C00kie-, 2021-01-30:
Hello. It did happen the two times I used it, for at least a few sec. Just the time to make a few moves on the games.

muesli, 2021-03-16:
Ran into the same situation yesterday with 0.14.0.

MFAshby, 2021-05-03:
I can't reproduce this on 1.0.0, drawing and 2048 game seem to work without crashing.

kieranc, 2021-10-01:
I also cannot reproduce, I believe this is resolved and can be closed.

Avamander, 2021-10-02:
Closing as it can't be reproduced.

closed 2021-10-02 by Avamander