InfiniTimeOrg/InfiniTime #176: Ability to dismiss notifications

Jab2870, 2021-01-22:
I have just received my pinetime, so it is very possible that I am missing
something, however, I can't find a way to dismiss notifications.

I am running version 0.10.0

Thanks in advance

JF002, 2021-01-22:
Hi and welcome among us :)

What do you mean by "dismiss the notification" ? Removing it from the list of notifications displayed in the notification app?

Jab2870, 2021-01-22:
Thank you.

Yes, that is what I was thinking

grayrattus, 2021-02-23:
I could try to implement this.

I was thinking about swipe right movement that could trigger a small **remove icon** which pops up from the right side of the screen and then second swipe right could remove the notification. A couple of swipes to the right could empty NotificationManager.

Swipe left would close **remove icon** and the user will not be able to delete notification till he swipes right again.

What do you think?

grayrattus, 2021-02-25:
FYI I started to work on that but I noticed that that first I need to learn a bit of LVGL :D 
I will still work on this feature but first I need to learn how to LVGL properly.

Is there a simpler way to preview/install changes than using the Gagetbridge app that installs my custom stuff? Currently, I'm putting built soft on so online hosting, then download it on phone, and from that, I install it on PineTime.

Avamander, 2021-02-25:
> Is there a simpler way to preview/install changes than using the Gagetbridge app that installs my custom stuff?

Yes, use a SWD programming dongle and use that to upload the firmware.

aymswick, 2021-12-22:
Is there currently no way to dismiss a notification? I've searched through a number of these issues which seem to indicate conflicting info about whether this is implemented or not. I've tried swiping left/right/up/down, long pressing, etc and can't remove a notification.

closed 2022-06-28 by JF002