InfiniTimeOrg/InfiniTime #179: Firmware downgrades to 0.7.1 on hard reset

Man-with-Arrow, 2021-01-23:
Using a sealed PineTime, which came with 0.7.1 (Jul 20 2020 build date). On hard reset, firmware version will revert to 0.7.1. This has happened with multiple newer versions, and after verifying the new firmware in the UI.

## Expected Behavior
Watch should retain current firmware version.

## Current Behavior
Watch reverts to factory-installed firmware.

## Steps to Reproduce

1. Unpair watch from devices.
2. Hold down button until display backlight starts alternating levels.
3. Let go of button, wait for watch to boot.
4. This isn't consistently reproducible, so you may have to repeat the above steps.

## Context (Environment)
Using a sealed PineTime. Not using the stock bootloader as far as I can tell; when I installed wasp-os a while back and reverted to MCUBoot, the boot logo stayed the same (i.e. a pinecone, instead of "INFINITIME" text).

Man-with-Arrow, 2021-01-23:
I'm an idiot, this is expected behaviour... See [here](https://lupyuen.github.io/pinetime-rust-mynewt/articles/mcuboot#manual-firmware-rollback-on-pinetime).

closed 2021-01-23 by Man-with-Arrow

Avamander, 2021-01-23:
That's a good thing to document though. Where did you look first when you encountered this behaviour?