InfiniTimeOrg/InfiniTime #180: Music integration on GB on 0.11.0 doesn't seem working

DanielSolmann, 2021-01-24:
I've updated my PineTime today, it was working previously on 0.10.0, but is broken now.

Avamander, 2021-01-24:
Not an InfiniTime issue, Gadgetbridge needs an update.

DanielSolmann, 2021-01-26:
Will it work in the next GB's release then? And does it work now, if I'll compile it by myself?

Avamander, 2021-01-26:
@DanielSolmann 
Depends on how fast the change is made on GB's side and if it's done before or after the next release.

JF002, 2021-01-27:
Yes, we changed the UUIDs of the BLE services for more coherency and, by doing this, we broke the music service implemented in Gadgetbridge and Amazfish. The last version of amazfish is already up to date, but I don't think anyone has already done the change in Gadgetbridge. 
I won't have the time to do it in the near future so I hope someone will do it for us :)

vbelloir, 2021-01-29:
With which version of Amazfish is it supposed to work?

With Infinitime 0.1.00, and amazfish 1.9.0, music integration is broken for me?

stephanlachnit, 2021-02-13:
I opened an issue for Gadgetbridge here: https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2199

stephanlachnit, 2021-02-14:
Fixed in https://codeberg.org/Freeyourgadget/Gadgetbridge/commit/b9647e26c3287053e88330ebc97ed25fafe29d7e

ncartron, 2021-02-16:
> With which version of Amazfish is it supposed to work?
> 
> With Infinitime 0.1.00, and amazfish 1.9.0, music integration is broken for me

@vbelloir I believe you had the answer in the meantime, but just for tracking purposes in that issue: the version of Amazfish wihch fixes that is 1.9.5

vbelloir, 2021-02-16:
> > With which version of Amazfish is it supposed to work?
> > With Infinitime 0.1.00, and amazfish 1.9.0, music integration is broken for me
> 
> @vbelloir I believe you had the answer in the meantime, but just for tracking purposes in that issue: the version of Amazfish wihch fixes that is 1.9.5

Well, not exactly. 
It's in 1.9.1 that music app is working. But this version wasn't release in openrepos for SFOS.

ncartron, 2021-02-16:
right, I forgot to mention that I was pointing to the downloadable version of Amazfish for SFOS :)

JF002, 2021-02-16:
@stephanlachnit Thank you very much for your work on Gadgetbridge :100: ! I was also missing the music integration since we've changed the UUIDs !

stephanlachnit, 2021-02-21:
@JF002 can you check the UUIDs or if it the latest Gadgetbridge works for you? Still doesn't work for me :/

stephanlachnit, 2021-02-21:
Nvm I'm stupid, I didn't look at https://github.com/JF002/Pinetime/commit/ba03372f3b99798de2ce497d34f9ee5697e3be40, only https://github.com/JF002/Pinetime/commit/ccbe2e8e1b15a09b4d9e51b54a5a2e4cdb847fdc. New PR incoming.

JF002, 2021-03-25:
Fixed in Gadgetbridge 0.55

closed 2021-03-25 by JF002